fougue/mayo #142: [ERROR]Save STEP file use  AP242_DIS schema

akjay, 2022-08-16:
Bother you again.
Since I made some GD&T definition with my shapes use DimTolTool, I want to save theme in my step files,so I use export function to save document,and manually change the default mode to AP242_DIS  use tools/options.After this has been done,run "export selected  items", but file can not be saved and an error named "File write problem" occur.
I use the file blow,just open it and change schema to AP242_DIS then do export can repeat this error.
[nist_ctc_05.zip](https://github.com/fougue/mayo/files/9348867/nist_ctc_05.zip)
[image]

HuguesDelorme, 2022-08-16:
Hello akjay
I could reproduce the problem as well on my side.
CAD Assistant is also failing to export your file with schema AP242DIS so it's likely a problem with OpenCascade itself.
I would suggest you report that issue in the OpenCascade forums.

akjay, 2022-08-16:
I try all schemas list in the options find out that all schema expect AP214_IS were failed, because the default schema is AP214_IS.So  I guess  something wrong with change "write.step.schema" parameter.
I note a comment said:

>     const int previousSchema = Interface_Static::IVal("write.step.schema");
>     rollback->change("write.step.schema", int(m_params.schema));
>     if (int(m_params.schema) != previousSchema) {
>         // NOTE from $OCC_7.4.0_DIR/doc/pdf/user_guides/occt_step.pdf (page 26)
>         // For the parameter "write.step.schema" to take effect, method STEPControl_Writer::Model(true)
>         // should be called after changing this parameter (corresponding command in DRAW is "newmodel")
>         m_writer->ChangeWriter().Model(true);
>     }

While with the param "rollback" destruct when step out it's scope the param rollback to its previous data,but  it seems that missing ChangeWrite().Model() call when change "write.step.schema" in destructor function in class "OccStaticVariblesRollback",so I comment the entire destructor function to disable rollback since it will change "write.step.schema" and may cause problem,and it works.

HuguesDelorme, 2022-08-16:
Interesting, you mean `STEPControl_Writer::Model(true)` has to be called in some way after "write.step.schema" was restored to its previous value by `OccStaticVariablesRollback` ?

akjay, 2022-08-16:
I mean that: 
follow the comment wrote in line 361 in file io_occ_step.cpp，once someone want to change the "write.step.schema" parameter, a STEPCotrol_Write::Model(true) must be called after the change to make it take effect,and the code does this in changeStaticVaribles function.And when the local variable leave its scope,the destruct function of class "OccStaticVariablesRollback" which OccStaticVariablesRollback::~OccStaticVariablesRollback() called,which make all the params changed befor restore to its previous value include "write.step.schema"  if changed,but a STEPCotrol_Write::Model(true)  call missing when resotre it, so I comment the entire function to disable restore action,and then save successed.
I do not know exactlly whether this is the bug that cause save error or not.

akjay, 2022-08-16:
Since export a document has two key steps in file io_occ_step.cpp,one is transfer and another one is writeFile.
After transfer function called, a restore will done when local variable "rollback" destruct but missing STEPCotrol_Write::Model(true) called with restore "write.step.schema",so next step writeFile may cause problem.
I add a skip statement in OccStaticVariablesRollback::~OccStaticVariablesRollback() function that:

> for (const StaticVariableRecord& record : m_vecRecord) {
>     if(record.strKey.compare("write.step.schema") == 0)
>        continue;
> ……
> }

HuguesDelorme, 2022-08-16:
@akjay Many thanks for investigating that issue.
To my understanding the problem is that `Model(true)` is called in `OccStepWriter::writeFile()` which clears what has been transferred previously(in the writer's transfer() function)
I've tested to avoid calling `Model(true)` in `OccStepWriter::writeFile()` and that solved the error
There's clearly a bug with `OccStepWriter` and I will provide a fix soon.

akjay, 2022-08-16:
So it's looks like the same way I solve this,since I didn't restore the schema parameter after transfer,the Model(true) will not be called in writeFile because the previous and current schema params is the same,will not step into if statement,and which will not clears what has been transferred brfore.

closed 2022-08-16 by HuguesDelorme