fougue/mayo #144: Error reading file on every step files

physnoct, 2022-08-24:


sebmestrallet, 2023-08-31:
Hello, I may not be related, but I had a similar issue (infinite loading) with only the first 3 lines of your logs.

My solution was to use Intel graphics instead of NVIDIA (on Ubuntu, with `__NV_PRIME_RENDER_OFFLOAD` and `__GLX_VENDOR_LIBRARY_NAME` shell variables).

sebmestrallet, 2023-08-31:
I tried to install from sources (I used the AppImage before), but sadly Qt 5.14 is not available on the Ubuntu 20.04 LTS apt repository. And I don't have the rights to add PPAs on my PC...

I forgot to say in my last message that the warning/error lines are still printed with Intel drivers but it works.

sebmestrallet, 2023-10-12:
Some news: I have another work PC on my hands, on Ubuntu 22.04 this time. The [build instructions for Debian](https://github.com/fougue/mayo/wiki/Build-instructions-for-Debian) work perfectly. I no longer have any problems with NVIDIA drivers. Here is the System Information output, OpenGL parts.

With Intel graphics drivers:
```
OpenGL:
    GLXDirectRendering: Yes
    GLXVendor: SGI
    GLXVersion: 1.4
    GLXClientVendor: Mesa Project and SGI
    GLXClientVersion: 1.4
    GLvendor: Intel
    GLdevice: Mesa Intel(R) Graphics (ADL GT2)
    GLversion: 4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
    GLSLversion: 4.60
    Max texture size: 16384
    Max FBO dump size: 4096x16384
    Max combined texture units: 192
    Max MSAA samples: 16
    Viewport: 100x100
    GPU memory: 31773 MiB
    MaxDegreeOfAnisotropy: 16
    MaxDrawBuffers: 8
    MaxClipPlanes: 8
    HasRayTracing: true
    HasRayTracingTextures: false
    HasRayTracingAdaptiveSampling: true
    UseVBO: true
    MaxDumpSizeX: 4096
    MaxDumpSizeY: 16384
    HasRayTracingAdaptiveSamplingAtomic: false
    HasTextureBaseLevel: true
    HasSRGB: true
    RenderSRGB: true
    IsWindowSRGB: true
    HasPBR: true
```

With NVIDIA drivers:
```
OpenGL:
    GLXDirectRendering: Yes
    GLXVendor: NVIDIA Corporation
    GLXVersion: 1.4
    GLXClientVendor: NVIDIA Corporation
    GLXClientVersion: 1.4
    GLvendor: NVIDIA Corporation
    GLdevice: NVIDIA T600 Laptop GPU/PCIe/SSE2
    GLversion: 4.6.0 NVIDIA 535.86.05
    GLSLversion: 4.60 NVIDIA
    Max texture size: 32768
    Max FBO dump size: 32768x32768
    Max combined texture units: 192
    Max MSAA samples: 64
    Viewport: 100x100
    GPU free memory: 3852 MiB
    GPU memory: 4096 MiB
    MaxDegreeOfAnisotropy: 16
    MaxDrawBuffers: 8
    MaxClipPlanes: 8
    HasRayTracing: true
    HasRayTracingTextures: true
    HasRayTracingAdaptiveSampling: true
    UseVBO: true
    MaxDumpSizeX: 32768
    MaxDumpSizeY: 32768
    HasRayTracingAdaptiveSamplingAtomic: true
    HasTextureBaseLevel: true
    HasSRGB: true
    RenderSRGB: true
    IsWindowSRGB: true
    HasPBR: true
```

I still have 
```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
```
warning lines printed at runtime in both cases.

HuguesDelorme, 2023-11-17:
Hello @sebmestrallet
Good to see it works on your side, thanks for the feedback
That TKOpenGl message is an OpenCascade warning and shouldn't cause rendering problems(see [comment](https://dev.opencascade.org/node/83575#comment-23084))

HuguesDelorme, 2023-11-17:
@physnoct 
Are you still having the issue? What about the 3D graphics, do you the STEP file correctly rendered?

alexmyczko, 2024-07-23:
meanwhile, i created an ITP and an official debian package is waiting at ftp-master new queue. you'll find a copy here http://bananas.debian.net/debian/mayo/git/

HuguesDelorme, 2024-07-24:
@alexmyczko 
Thanks for your effort, notice that some debian package was already published at xtradeb:
https://xtradeb.net/apps/mayo

HuguesDelorme, 2026-03-24:
No recent activity and should be fixed with #380