fougue/mayo #149: It does not run on Linux Mint 19.3

mnesarco, 2022-09-09:
Hello, I get this error on Linux Mint 19.3:

```
➜  ./Mayo-0.6.0.1012-e55c5ee-x86_64.AppImage
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKBRep.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKernel.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKG3d.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKGeomAlgo.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKGeomBase.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKMath.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKMesh.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKOpenGl.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKService.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKV3d.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKSTEP.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libQt5Widgets.so.5)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libQt5Gui.so.5)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libQt5Core.so.5)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libQt5Core.so.5)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libTKBO.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libfreeimage.so.3)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libpng16.so.16)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libicui18n.so.66)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libicuuc.so.66)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libglib-2.0.so.0)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libopenjp2.so.7)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libraw.so.19)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libtiff.so.5)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libwebpmux.so.3)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libwebp.so.6)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/libIlmImf-2_3.so.24)
/tmp/.mount_Mayo-0tf5RTA/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Mayo-0tf5RTA/usr/bin/../lib/liblcms2.so.2)

```

HuguesDelorme, 2022-09-12:
Hello @mnesarco
Linux Mint 19.3 was released on december 24, 2019 and Mayo's AppImage was created on Ubuntu 20.04 (april 2020 then) hence the glibc mismatch.

Is there any chance you can install a newer version of Linux Mint ?

HuguesDelorme, 2022-09-12:
See also https://github.com/fougue/mayo/issues/26#issuecomment-1243411527

mnesarco, 2022-09-12:
Do you mean mint 19.x and ubuntu 18.x are not supported?
It is ok if not supported, but it would be great to have this info in the readme.

Thanks.

HuguesDelorme, 2022-09-20:
Added notice _"Mayo AppImage for Linux was built on Ubuntu 20.04 so trying to use the application on older distributions might fail"_ in the release [page](https://github.com/fougue/mayo/releases/tag/v0.6.0)

closed 2022-09-20 by HuguesDelorme