fougue/mayo #151: Error on Build ( MacOS)

KeithSloan, 2022-09-10:
In file included from ../mayo/src/base/application.cpp:7:
In file included from ../mayo/src/base/application.h:9:
../mayo/src/base/application_ptr.h:9:10In file included from :../mayo/src/base/application_item.cpp :7:
fatal errorIn file included from : ../mayo/src/base/application_item.h:9:
In file included from ../mayo/src/base/document.h:9:
../mayo/src/base/application_ptr.h:9:10: fatal error: 'TDocStd_Application.hxx'
      file not found
'TDocStd_Application.hxx'
      file not found
#include <TDocStd_Application.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~
#include <TDocStd_Application.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [application_item.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
make: *** [application.o] Error 1
keithsloan@Keiths-iMac-2 mayo_build

HuguesDelorme, 2022-09-12:
Hello @KeithSloan
What steps did you follow to build Mayo ?
Have you looked at https://github.com/fougue/mayo/wiki/Build-instructions-for-macOS ?

grexe, 2022-09-13:
same error here on both `develop` and `master` branches.
On Mac ARM64, followed the instructions, Qt and Cascade installed and detected, the file is indeed not there.

```
fatal error: 'TDocStd_Application.hxx' file not found
```

HuguesDelorme, 2022-09-14:
Hello again @KeithSloan and @grexe 
The build instructions for macOS were fixed(see the updated [wiki](https://github.com/fougue/mayo/wiki/Build-instructions-for-macOS))
Run these commands:
````
cd mayo-build
CASCADE_BASE_DIR=`brew info opencascade | grep -E -i --only-matching --max-count=1 "^(/[a-z\.\-_0-9]+)+"`
qmake ../mayo CASCADE_INC_DIR=$CASCADE_BASE_DIR/include/opencascade CASCADE_LIB_DIR=$CASCADE_BASE_DIR/lib
make -j2
````
The GitHub CI for macOS was also fixed(see [action](https://github.com/fougue/mayo/actions/runs/3051253177/jobs/4919264745))

Let me know if that fixes the issue on your side so we this issue can be closed

grexe, 2022-09-16:
ok the build runs through now and I can launch the application (app icon is blank though).
However when trying to load a sample STEP file from [nist.gov](https://www.nist.gov/ctl/smart-connected-systems-division/smart-connected-manufacturing-systems-group/mbe-pmi-0), namely [this one](https://www.nist.gov/document/nist-cad-model-ftc-06), and opening `nist_ftc_06_asme1_rd.stp`, I only get a black canvas.

The same file opens fine with e.g. CAD Assistant, which also uses OpenCascade.

HuguesDelorme, 2022-09-19:
Hello @grexe
Thanks for your feedback
Discussion about runtime problem goes on in #155

closed 2022-09-19 by HuguesDelorme