fougue/mayo #152: Installer closes all open programs and tries to force a restart

dignakov, 2022-09-11:
The app looks great, but as it was finishing installing it closed everything and tried rebooting the computer without my input or confirmation (Windows installer). This could cause people to loose work.

HuguesDelorme, 2022-09-12:
Hello @dignakov 
That's weird I've never experienced that behavior when testing Mayo's InnoSetup installer within a Win7 virtual machine.
There's also no such instruction in the InnoSetup script to close all apps and request reboot.
Maybe that's caused by the MSVC redistributable package which is executed at the end of installation.

dignakov, 2022-09-12:
That could be it. I can try re-installing now that I've the MSVC pack there already. Also, should have been more specific, this is on Win 10.

Let me know if you want me to try anything else.

HuguesDelorme, 2022-09-12:
Hi again @dignakov
Yes that would be relevant to try install Mayo again and check if reboot is still required.
Thanks!

HuguesDelorme, 2023-01-12:
No update

closed 2023-01-12 by HuguesDelorme