fougue/mayo #155: No 3D rendering(black area) on macOS

HuguesDelorme, 2022-09-19:
Reported by @grexe in issue #151

> However when trying to load a sample STEP file from nist.gov, namely this one, and opening nist_ftc_06_asme1_rd.stp, I only get a black canvas. The same file opens fine with e.g. CAD Assistant, which also uses OpenCascade.

Notice that some other user(@livigni) mentionned Mayo to run correctly on macOS(see https://github.com/fougue/mayo/issues/67#issuecomment-1137480691)

HuguesDelorme, 2022-09-19:
@grexe
Is there any OpenGL error prompted in the terminal when running Mayo ?
Can you please post here Mayo's console output on your machine ?

As a workaround you can comment the `if()` block between lines 350-363 in `src/app/main.cpp` : https://github.com/fougue/mayo/blob/e41d872be90744e9c8efd890e163f09f1d7c8c01/src/app/main.cpp#L350
Build and run again

zambetti, 2023-02-02:
Hello @HuguesDelorme & @grexe

I was experiencing this same issue on MacOS 13.0.1 (22A400), I followed your advice to comment out the block in main.cpp under the comment "// Use QOpenGLWidget if possible"

All is working fine now. Looking good.

jhillhouse92, 2023-03-07:
@HuguesDelorme comment fixed it. Still issue as of Mar 6 23.

HuguesDelorme, 2023-03-07:
@zambetti and @grexe
Please try new option [Force OpenGL fallback widget](https://github.com/fougue/mayo/issues/180) to tweak OpenGL support.

andrsd, 2023-03-30:
Hi!

I had the same issue with the black screen. I tried the fallback widget option (i.e. I turned it off and restarted `mayo`), but that did not work for me. Commenting out the whole block in `src/app/main.cpp`, i.e.:
```
#if OCC_VERSION_HEX >= 0x070600
...
#endif
```
(roughly lines 392-408) fixed my problem. Now, I can see the model.

I am on Ventura 13.1, Apple M1 chip, built `mayo` against Qt 6.2.4 (IDK if that matters).

HuguesDelorme, 2023-03-31:
@andrsd 
The "fallback widget" option should be set to ON.
Please check again with that option turned ON(and without the comment of course)
Let me know your feedback!

andrsd, 2023-03-31:
The fallback widget was originally ON. I believe that's the default (?). That's when I had the issue with black screen, so I went  to look for the cause and found this issue. I can try again.
I forgot to mention I built the this version: c85fe27fbfa27ba9b163b01bb54036473fa31e4f

andrsd, 2023-03-31:
I tried a clean build of c85fe27fbfa27ba9b163b01bb54036473fa31e4f, made sure the `Fallback widget` was `ON` and all works as expected. Hmmm... 🤔

HuguesDelorme, 2023-03-31:
@andrsd
I confirm the `forceOpenGlFallbackWidget` option is set to ON for macOS by default.
https://github.com/fougue/mayo/blob/c85fe27fbfa27ba9b163b01bb54036473fa31e4f/src/app/app_module_properties.cpp#L87

I don't like this option, it's preferrable to fix the OpenGL framebuffer rendering with Qt for macOS
But until this gets fixed, seem like for macOS this workaround is required.

HuguesDelorme, 2023-04-01:
@andrsd 
You might have spotted a potential issue, I need to check this.
The `forceOpenGlFallbackWidget` setting is explicitly loaded here:
https://github.com/fougue/mayo/blob/925403f13a3c5f6642c4f3812000ea4b8551514c/src/app/main.cpp#L394
But for some reason this could fail on the very first execution of Mayo

gwonhong, 2025-01-03:
this might not be so helpful, but I also experienced the very same bug: blank screen **only at first execution** (with .stp file), and **runs fine from the next execution**.

detailed info of my machine is as follows:
Apple M1 Pro
macOS 15.2 24C101 arm64