fougue/mayo #156: Add to package managers like scoop, winget and chocolatey

Gaweringo, 2022-10-01:
I think it would be nice if this tool would be made available through some Windows package managers like [scoop](https://github.com/ScoopInstaller/.github/blob/main/.github/CONTRIBUTING.md#for-scoop-buckets), [winget](https://learn.microsoft.com/en-gb/windows/package-manager/package/repository) and [chocolatey](https://docs.chocolatey.org/en-us/#create-packages).

Adding mayo to these package managers would make it (imo) a lot easier to install and update.

HuguesDelorme, 2023-08-28:
@Silther
Yes seems to be added with PR https://github.com/microsoft/winget-pkgs/pull/95480

HuguesDelorme, 2023-09-13:
Mayo 0.6.0 and 0.7.0(latest at the time of writing) versions are now available in winget-pkgs, see merged pull requests:
https://github.com/microsoft/winget-pkgs/pull/95480
https://github.com/microsoft/winget-pkgs/pull/119418

See also https://winstall.app/apps/Fougue.Mayo

Mannshoch, 2025-03-24:
I detected that winget do find Mayo 0.8.0 and after install it get the problem to determine the installed mayo version.

```
Name        ID          Version Verfügbar Quelle
------------------------------------------------
Mayo v0.8.0 Fougue.Mayo Unknown 0.8.0     winget
```

Cause of that I missed the new version 0.9.0

HuguesDelorme, 2025-04-09:
> I detected that winget do find Mayo 0.8.0 and after install it get the problem to determine the installed mayo version.
> 
> ```
> Name        ID          Version Verfügbar Quelle
> ------------------------------------------------
> Mayo v0.9.0 Fougue.Mayo Unknown 0.8.0     winget
> ```
> 
> Cause of that I missed the new version 0.9.0 [Edit] it appears the same after install v0.9.0

@Mannshoch  That's weird because https://winstall.app/apps/Fougue.Mayo clearly states that v0.9.0 is latest version