fougue/mayo #158: GLTF export coordinates converter

Daniel4144, 2022-10-20:
Hi,

I imported a step file and exported it as gltf/glb and the resulting model is oriented wrong (tested in different viewers - Windows 3D viewer, Blender and in Unity imported with [glTFast](https://github.com/atteneder/glTFast).
I tested with the export setting Coordinates Converter which, as I understand, should result in different orientations, but the three options (Undefined, negZfwd_posYup, posYfwd_posZup) seem to make no difference.
I also tried to export as obj, which has the same setting, with the same result.

Am I doing something wrong or is the coordinate conversion not working?

HuguesDelorme, 2022-11-09:
Hello @Daniel4144
Thanks for reporting this issue, sorry for the late response.
I've provided a fix in `develop` branch : glTF and OBJ writers now have parameters for input/output coordinate systems
Let me know if it fixes the issue on your side

Daniel4144, 2022-11-10:
Thank you for looking into the issue @HuguesDelorme
Is there a build with the fix so I can check if it is fixed on my side?

HuguesDelorme, 2022-11-10:
No unfortunately, not currently.
However I can generate one and give you a link to it.
What OS are you using ? Windows ?

Daniel4144, 2022-11-10:
That would be great, I really appreciate your help.
I am using Windows 10

HuguesDelorme, 2022-11-10:
Here it is:
https://www.fougue.pro/share/bin/mayo_v0.7.0.1043-77f594c_winx64_installer.exe
Let me know your feedbak

Daniel4144, 2022-11-11:
I can confirm that this version works for me - now I am able to adjust the coordinate system, tested with both gltf and obj export.

Remark:
I don't see the benefit of having two parameters for the coordinate system. Maybe I am missing something, but isn't the input coordinate system always the same (the viewer always uses z up)?
Imo the default value for the input coordinate system should at least be posZup, in the test build you provided the default was undefined (and it crashed when I tried to restore default values).

HuguesDelorme, 2022-11-14:
Thanks for your feedback.
I'm also doubtful about whether input coordinate system parameter is useful.
Example, here is a base part:
[image]

Export posZup -> posYup:
[image]

Export posYup -> posZup:
[image]

The input coordinate system parameter allows another transformation, but is it relevant ? I would say yes but not 100% sure

closed 2023-01-12 by HuguesDelorme