fougue/mayo #160: Can running on web!?

long0419, 2022-11-07:


HuguesDelorme, 2022-11-07:
There is a plan in the mid-term to make Mayo run on the web by "porting" it to WebAssembly

closed 2022-11-07 by HuguesDelorme

long0419, 2023-08-19:
The Mayo application has encountered several performance issues on the Windows client, and in some cases, it's even incompatible after installation. This client is primarily intended for 3D file format previewing, including interactions with 3D files. WebAssembly (wasm), aside from being cross-platform and independent of the underlying operating system, offers excellent zero-copy functionality and performs well in file rendering.

@HuguesDelorme I can contribute to this project if possible. Rewriting it in Rust could also be a good option.