fougue/mayo #162: Functions about STEP to display

Gentle-Liu, 2022-11-11:
Excuse me, I have a few questions and would like your reply.
1. Regarding the Measure shapes section, the mouse selects the graphics to highlight and gives the parameters. For example, if you select a cylindrical surface to highlight and display the area, which function is the interface for this function?
2. Is highlighting and giving parameters calculated at the display layer or extracted from the STEP file? Is there an interface for extracting information from STEP files to display layer?
3. How can I open this file given the STEP file path and file name?
Thanks a lot.

HuguesDelorme, 2022-11-12:
1. Sorry I don't undertand this question, what do you mean by "which function is the interface for this function" ?
2. Mayo uses the underlying geometry/BREP data of the selected 3D entities to compute the measurements(it doesn't work for mesh file formats)
3. Sorry I don't undertand this question

Gentle-Liu, 2022-11-13:
Thank you very much for your answer!

closed 2022-11-14 by HuguesDelorme