fougue/mayo #163: DWG

calumk, 2022-11-11:
Support DWG import

HuguesDelorme, 2023-05-02:
DWG file format specification:
https://www.opendesign.com/files/guestdownloads/OpenDesign_Specification_for_.dwg_files.pdf

calumk, 2024-02-02:
This is still a huge need for me -> I have hundreds of .dwg files generated by [pcon](https://pcon-planner.com/en/) that i would love to get into other formats for rendering in game engines and online

HuguesDelorme, 2024-02-02:
Yes supporting dwg support in Mayo would be great. In the recent past I took some time to look at [libredwg](https://github.com/LibreDWG/libredwg) library. It's a very difficult library to use, there's not that much documentation and examples of projects based on it.
In the meanwhile you can convert dwg files to dxf with dwg2dxf from libredwg. Is it relevant for you?

calumk, 2024-02-02:
At the moment I can do this manually by opening and converting using a cad application and exporting, it's just that it's windows only, and a bit labour intensive 

Libredwg can get me part of the way, but stringing together existing tooling is a bit of a pain, ideally a single cross platform app to do this would be ideal 👍

Mannshoch, 2024-09-06:
If you need some test files, I created a set of dxf and dwg files for inkscape which support import of 2D Data:
https://gitlab.com/inkscape/extensions/-/merge_requests/629#note_2031419407

this test files contains also some 3D bodies.