fougue/mayo #166: [BUG] Problems exporting from imported PLY, IGES, STL file formats

Abde5, 2022-11-24:
Hello,

Thanks again for the active development of Mayo. It's crazy good.

Whenever I import a PLY, IGES, STL file in Mayo, I am not able to export them to any format. I tried exporting as GLTF, which gives a 1KB file with no data, and when exporting as an OBJ, there is a "writing problem" error.

The importing works well, is only the exporting to other formats where I am encountering problems.

HuguesDelorme, 2022-11-28:
Hello @Abde5
Thanks for reporting the issue, I could reproduce it.
However exporting IGES "data" to glTF, STL, OBJ, ... works fine on my side

Abde5, 2022-11-28:
Hello @HuguesDelorme

Thanks for answering the issue. You are right, this behaviour happens only in poor-written iges files. I tried with a better IGES file and it seems to work perfectly.

INPUTsys-Chris, 2022-12-09:
Hi,

for me there is always a "File transfer problem" when exporting while no node is selected.
Maybe this could be related to some problems you see. Otherwise it's a separate issue.
Then I could open a new issue if you like.

HuguesDelorme, 2022-12-09:
Hello @INPUTsys-Chris
Yes that's a different issue. When no Model tree nodes were selected then there's nothing to export and Mayo reports a somewhat confusing error.
That would be kind you file create a different bug report for this one, thanks!