fougue/mayo #177: An error occurred when starting the program

piggycatc, 2023-02-22:
When I run it for the first time and open the file, it works fine, but when I run the program again later, the program closes.
It appears to be due to another language on the path.
If there are other languages ​​on the path, the program will exit without running.

HuguesDelorme, 2023-02-22:
Can you give an example of a file path that causes the crash ?

akjay, 2023-02-22:
If the path has chinese characters，it will crash.

HuguesDelorme, 2023-02-22:
@akjay
Can't reproduce it with develop branch:
[image]

Are you using v0.6.0 binaries provided in the Releases page?

akjay, 2023-02-23:
> @akjay
> Can't reproduce it with develop branch:
> [image]
> 
> Are you using v0.6.0 binaries provided in the Releases page?

I use 0.6.0.1012.
After the lastOpenFolder record this file path，the you can try open Mayo.exe and close it with nothing other operation，maybe two or three times it will crash.Then need go to regedit to clean lastOpenFolder setting for recover.

piggycatc, 2023-02-23:
Just deleting the 'lastOpenFolder' registry will make it run normally again.

And the list of recently opened files is still visible.

HuguesDelorme, 2023-03-13:
@piggycatc and @akjay 
Just tried the v0.6.0.1012 version within a virtualized Win7 x64 machine:
[image]

Still can't reproduce the problem!
Maybe related to the locale of your systems?

piggycatc, 2023-03-14:
It's not a filename problem, it's a path problem. If there are other characters on the path, the program will stop. Or you can check it by running it and turning it off three times.

windows10  x64, i7 - 16G - Nvidia gtx1050
windows11 x64 i5 - 16G - Nvidia rtx3050ti

HuguesDelorme, 2023-03-14:
@piggycatc and @akjay
Here is what I tried without any problem:
[image]

The full path of the file is `C:\Tests\热情\file_一号文件.stp`
Should this path cause a crash on your machine?

HuguesDelorme, 2023-03-29:
@piggycatc and @akjay 
Please let me know if the issue still remains with new 0.7.0 version
Thanks!

piggycatc, 2023-03-30:
> 
Tested with version 0.7.0
When I open the file for the first time, it runs normally.

The text on the path is normal Korean.

[image]

It is an Lastopen folder's Path in the registry

[image]

third run and then program stopped 

[image]

text broken in the registry path
[image]

HuguesDelorme, 2023-03-30:
Maybe I found the explanation:
https://engineertips.wordpress.com/2021/10/03/windows-path-separators/

HuguesDelorme, 2023-06-01:
@piggycatc @akjay 
Can one or both of you please join here a crashdump file?
Make sure crashdump generation is ON with your system: https://learn.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps

Thanks for your feedback!

akjay, 2023-06-13:
> @piggycatc @akjay 
> Can one or both of you please join here a crashdump file?
> Make sure crashdump generation is ON with your system: https://learn.microsoft.com/en-us/windows/win32/wer/collecting-user-mode-dumps
> 
> Thanks for your feedback!

ok,will try later.

kantoku-code, 2023-08-14:
@HuguesDelorme -San.

Thanks for the great software.

I am Japanese, but similarly, if the file path contains 2 byte characters (multibyte characters), it crashes and does not start when restarting.
This is ver 0.7.0.

HuguesDelorme, 2023-11-02:
@kantoku-code 
Can you checkout `develop` branch and check if the crash still happen? It might be fixed as as side effect of #219

kantoku-code, 2023-11-02:
@HuguesDelorme -San.

We decided to respond by creating a VBScript to start Mayo after deleting the registry.
[MayoStarter.zip](https://github.com/fougue/mayo/files/13241387/MayoStarter.zip)

HuguesDelorme, 2023-11-14:
Duplicate of #219

closed 2023-11-14 by HuguesDelorme

HuguesDelorme, 2023-12-20:
@kantoku-code @akjay @piggycatc 
Can one of you please test if this crash still happen with latest version([v0.8.0](https://github.com/fougue/mayo/releases/tag/v0.8.0))?

kantoku-code, 2023-12-21:
@HuguesDelorme  -San.

Sorry for the delay in confirmation.
For Japanese, this issue has been resolved in v0.8.0.
Thanks.