fougue/mayo #178: on wayland: "Error: OpenGl_Context is unable to wrap OpenGL context"

blackvladimir, 2023-02-22:
If i run mayo on wayland it starts fine but as soon the 3d view opens (ie. opening file or creating new document) it outputs error to terminal, 3d view is black, and then it crashes after any interaction. 

In terminal it looks like:
```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
INFO: OpenGL v4.6
Error: OpenGl_Context is unable to wrap OpenGL context
Segmentation fault (core dumped)
```

I am running it on swaywm on archlinux. I tried latest master and develop branch with same result.

When I run it in X11 it works fine

jhillhouse92, 2023-03-07:
I have similar issue with OS X Big Sur. 3d view shows black screen and debug message is:

```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
INFO: OpenGL v2.1
Mesh /Users/JHILLHOUSE/Downloads/cube.obj
[8 nodes] [12 2d elements]
[read in 0.000803 s]
````

HuguesDelorme, 2023-03-07:
@blackvladimir and @jhillhouse92 
Please try new option [Force OpenGL fallback widget](https://github.com/fougue/mayo/issues/180) to tweak OpenGL support.

blackvladimir, 2023-03-07:
With  "Add option to force OpenGL fallback widget" it ends with:
```
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0xb
  Serial number of failed request:  29
  Current serial number in output stream:  30
```

HuguesDelorme, 2023-03-08:
@blackvladimir
Thanks for your feedback, that's very valuable. I have to take time in the short-term to properly investigate the issue.
In the meanwhile, please try this workaround(when Wayland is "active") by forcing `xcb` platform with:
`$> mayo --platform xcb`
Or
`$> QT_QPA_PLATFORM=xcb  mayo`

Let me know if it works under Wayland, also try with the "Force OpenGL fallback widget" option ON/OFF

blackvladimir, 2023-03-09:
With the `--platform xcb` it works.  "Force OpenGL fallback widget" can be ON or OFF and I don't see difference.

blackvladimir, 2023-03-09:
> Thanks again for your feedback, I will prepare a patch to force `xcb` platform on Unix-like systems.
> When "Force OpenGL fallback widget" is OFF does it allow the widget panels to be translucid other the 3D view(eg measure tool widget)?

No it is solid.
[image]

When "Force OpenGL fallback widget" is ON it is slightly different color

[image]

HuguesDelorme, 2023-03-13:
After investigation it appears that OpenCascade TKOpenGl depends on XLib for Linux or more generally when XLib was detected at compilation time. See for example implementation of `Aspect_DisplayConnection` where XLib is explicitly used with `#if defined(HAVE_XLIB) ... #endif` conditions.
So there's no chance to support Wayland natively with OpenCascade in its current status(latest version is 7.7.0 at the time of writing).
The best workaround so far is to force `xcb` QPA platform at Mayo startup time so X11 is used.

closed 2023-03-13 by HuguesDelorme