fougue/mayo #184: Error when exporting to obj

swall-pxc, 2023-03-24:
I want to convert stp files to obj.

So I built a Docker container, from a simple Debian base image, following the instructions in the wiki.
However running:
```bash
mayo model.stp --export model.obj
```
Returns:
```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
Error during export to 'model.obj' No supporting writer
100% Imported
```

Do I need to provide anything else so that obj can be exported?
Any Advice would be appreciated and let me know if you need more information.

HuguesDelorme, 2023-03-24:
What version of OpenCascade are you using?
Notice obj export requires OpenCascade >= 7.6.0
This is stated in this [wiki page](https://github.com/fougue/mayo/wiki/Supported-formats)

swall-pxc, 2023-03-24:
Ok I was using 7.5 updated it and now it works. Thank you :)

closed 2023-03-24 by swall-pxc