fougue/mayo #196: stp file cannot displayed in VS2017.

huangyangpi, 2023-05-30:
Thank you for providing the code. When I load the .stp file, it cannot be displayed. My environment is VS2017+QT5.13.0+opencascade 7.4.0 Do you know what the reason is?

HuguesDelorme, 2023-07-13:
Mayo should prompt a warning at startup(and also at QMake build step) if the detected Qt version is considered too old

HuguesDelorme, 2023-07-13:
See #211 for tracking the issue

closed 2023-07-13 by HuguesDelorme

CreativeLau, 2024-12-04:
> Thank you for providing the code. When I load the .stp file, it cannot be displayed. My environment is VS2017+QT5.13.0+opencascade 7.4.0 Do you know what the reason is?

I encountered the same problem. 
Environment: VS2017+Qt5.12.10+occt7.8.0
Finally I found the reason.

It is related to src/qtbackend/qt_signal_thread_helper.cpp
Here we use this method to call the slot function.
`QTimer::singleShot(0, std::any_cast<QObject*>(context), fn); `
Finally will call QTimer::singleShotImpl function.

In Qt5.14 QTimer::singleShotImpl will call QMetaObject::invokeMethodImpl when the variable msec is zero.
But in Qt5.13/5.12 QTimer::singleShotImpl will new QSingleShortTimer directly. This cause the timer started in thread, and the slot function could not run.
To fix it for Qt5.13/5.12, we can call QMetaObject::invokeMethod directly here. This will make the slot function run in main thread queued.
`QMetaObject::invokeMetho(std::any_cast<QObject*>(context), fn, Qt::QueuedConnection);`

I think this modification can be considered to be applied whether if you consider compatibility with lower versions of Qt or not.

HuguesDelorme, 2024-12-04:
Hello that's great info, thanks!
Does Mayo with Qt5.12 works fine with your patch?
If yes then feel free to provide a pull request for this issue
That would be really appreciated

CreativeLau, 2024-12-05:
Hi @HuguesDelorme
Mayo works fine with the patch under Qt5.12.
In order to be compatible with Qt5.12, there are 3 additional patches should be applied.
1. Change Qt version check from 5.14 to 5.12 in CMakeList.
2. In [ src/app/qtgui_utils.cpp](https://github.com/fougue/mayo/blob/4fdc1cc778876921ad5124902957f58586adcdb8/src/app/qtgui_utils.cpp) method toQPixmap. Since Qt5.12 does not support QImage::Format_BGR888. I consider blocking the case of Image_Format_BGR for Qt5.12 by adding a version check macro.
3. In [src/app/widget_main_control.cpp](https://github.com/fougue/mayo/blob/4fdc1cc778876921ad5124902957f58586adcdb8/src/app/widget_main_control.cpp) method reloadDocumentAfterChange.  Since Qt5.12 does not support Qt::MarkdownText. I consider using Qt::AutoText for Qt5.12 by adding a version check macro.

Should I apply this changes in the pull request.

HuguesDelorme, 2024-12-05:
Hi @CreativeLau these patches sound great, thanks for the effort!

CreativeLau, 2024-12-05:
I have submitted pull request  #309