fougue/mayo #197: Help! Files can be opened,but do not show

HonorableLeon, 2023-05-31:
Hi my friend!Thanks for your codes!
We are building the project with **Microsoft Visual C++ 2019** + **OpenCASCADE-7.7.0-vc14-64** + **Qt 5.15.2** in Windows 10,after some times trying,we succeed and the program runs.
However，when we open files,the programe shows nothing.
So,can you help?
[image]

HuguesDelorme, 2023-05-31:
Hello @HonorableLeon
Are you using prebuilt Qt 5.15.2 binaries or did you build it yourself?
There are no icons in your screenshot, Mayo icons are in svg format and require `Qt5Svg.dll` + plugins `qsvgicon.dll` and `qsvg.dll`
So check you have those DLLs and they are accessible to Mayo exe

Also it's worth trying to download and install Mayo 0.7.0 to check it works on your system, see https://github.com/fougue/mayo/releases/tag/v0.7.0

HonorableLeon, 2023-06-01:
Thanks for your answer@HuguesDelorme 
I'm using the prebuilt Qt 5.15.2 binaries,these dlls are copied in the same folder with Mayo exe,but the programe shows blank ,same with yesterday.
Mayo 0.7.0 was working on my system.

HonorableLeon, 2023-06-01:
Solved!Works!Thank you my friend @HuguesDelorme 
[image]
My settings:
[image]

closed 2023-06-01 by HonorableLeon