fougue/mayo #198: build error

thyis, 2023-06-01:
Dear sir,
When I copy the compiled project from one computer to another computer, compiled it  and appears "
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(209,5): error MSB6006: “cmd.exe”, Have quit，code is 2"。 It's been bothering me for days.

HuguesDelorme, 2023-06-01:
Hello @thyis
You should run the full build procedure on the other computer as described in https://github.com/fougue/mayo/wiki/Build-instructions-for-Windows-MSVC
If that's not satisfying, then describe more your objective

thyis, 2023-06-02:
Year, it's  OK running the full build procedure.   but  it could be copied directly to other computers and compiled  before version 7.0.

closed 2023-06-02 by thyis