fougue/mayo #203: color face

thyis, 2023-06-25:
Hello, sir.
When I colored a face of the imported stp, it didn't work.  Export and re-import the colored stp, the whole stp color is OK but the face not.  What should I do to color the face? 
Looking forward to your reply.

HuguesDelorme, 2023-06-26:
Hello @thyis 
It's hard to help you precisely without knowing what you did
Is there a branch/commits where I can see your changes?

thyis, 2023-06-27:
Thanks.  the face of .stp has been successfully colored, but not the edge.
[image]

HuguesDelorme, 2023-06-27:
You probably need to create an XDE subshape for the edge to be colored and set the color with:
`doc->xcaf().colorTool()->SetColor(edgeLabel, color, XCAFDoc_ColorCurv);`

thyis, 2023-06-27:
it didn't work, like this:
if (gfxShape.ShapeType() == TopAbs_ShapeEnum::TopAbs_EDGE)
				{
					TDF_Label gfxlabel = shapeTool->FindMainShape(gfxShape);
					TDF_Label colorEdge = doc->xcaf().shapeTool()->AddSubShape(gfxlabel, TopoDS::Edge(gfxShape));
					doc->xcaf().colorTool()->SetColor(colorEdge, color, XCAFDoc_ColorCurv);
				}

thyis, 2023-06-28:
Thanks.   it work.
[image]

thyis, 2023-06-28:
How it find the unique identification of the edge, such as a number of integer. Like Unigraphics NX, every point, edge, line, and surface has a six-digit ID,

thyis, 2023-07-05:
Dear  sir,
it surprised change "object->Attributes()->SetFaceBoundaryDraw(false)" to  make edge coloring work.
I want to give each edge or face unique ID,  but  it failed,

HuguesDelorme, 2023-07-13:
What is not working about topology unique ID?
I mean what is the problem with the code you mentionned above?

closed 2023-07-25 by HuguesDelorme