fougue/mayo #205: Smaller obj file size?

swall-pxc, 2023-07-05:
# Some Background info:
- I use the cli in order to convert stp files to obj files in a pipeline.
- Later I want to ship the resulting obj files in a Web App.
- I would like to have about 10x smaller files (to speed up loading etc.)
- I am fine with worse quality

# Questions:
- Is there some parameter to reduce the file size (lowering the quality)?
- Is there something in the source code to change that at build time?

HuguesDelorme, 2023-07-05:
Hello @swall-pxc 
Have a look at the "BRep Meshing" parameters in Tools->Options dialog:
[image]

Try to downgrade the mesh quality from "Normal" to "Coarse" or "Very Coarse"
Once you find the values that meet your needs then export the settings with:
[image]

You will get an INI file that you can pass to mayo CLI:
`$> mayo.exe D:\Data\inputfile.step --settings myoptions.ini --export outputfile.obj`

You can edit the INI file in a text editor and change the values directly, name of the relevant parameter is `meshingQuality`

HuguesDelorme, 2023-07-05:
@swall-pxc 
See also related issue #126

closed 2023-07-05 by swall-pxc