fougue/mayo #213: getShape is null

thyis, 2023-07-24:
Hello：
Dear sir:
when  i pick the ais1_oc_214.stp,  getShape  is null  ,what happened.
[image]

HuguesDelorme, 2023-07-24:
Hello, it's hard to give you some clues about your issue.
Seems like the graphics owner is no longer valid after the shape move.
Is the code of `getShape()` the same as in `measure/measure_tool_brep.cpp` at L104?

thyis, 2023-07-24:
yes  the code of getShape()  is the same as in measure/measure_tool_brep.cpp at L104

thyis, 2023-07-24:
Is it possible that it is caused by the assembly relationship， instance label？

closed 2023-07-25 by HuguesDelorme

thyis, 2023-07-26:
How to update synchronously  owner been modified shape?