fougue/mayo #214: export  then import  location  make reference

thyis, 2023-07-27:
Hello,
when  i  choose  a shell  moving ,  
1.export it , 
2. import it,
3. it could changes location,  but creates reference with location to it,
[image]
[image]

HuguesDelorme, 2023-07-27:
Hello @thyis
I'm pleased to help you with your software derived from Mayo
But the "Issues" pages are dedicated to Mayo itself, reporting your issues specific to your software might confuse Mayo users
Please create a new thread in the "Discussions" place. I suggest the "Show and tell" category(see https://github.com/fougue/mayo/discussions/categories/show-and-tell)
In the meanwhile I'm going to close that issue, see you in the Discussions soon!

closed 2023-07-27 by HuguesDelorme

thyis, 2023-07-27:
i am so sorry for the mayo user.