fougue/mayo #216: [FIX] correct build instructions

jons-pf, 2023-07-27:
Hi there,

first a big thanks for this nice software!

I think I spotted a typo in the [build instructions for Debian](https://github.com/fougue/mayo/wiki/Build-instructions-for-Debian).
Probably the `qmake` step has to refer to `..` from within the `mayo-build` directory and not `../mayo`?
I therefore suggest to edit the `qmake` line to the following:
`qmake .. CASCADE_INC_DIR=$CASCADE_INC_DIR CASCADE_LIB_DIR=$CASCADE_LIB_DIR`

What do you think?

HuguesDelorme, 2023-07-27:
Hello
Not sure if `qmake ..` is more correct than `qmake ../mayo`
Actually the real confusing thing is about the "assumed" current directory in the build instructions.
It should be stated!

HuguesDelorme, 2023-09-04:
The build instructions for Debian are correct: after the `git clone ...` step a sub folder named `mayo` is created inside the current directory. Next instruction to follow is `mkdir mayo-build && cd mayo-build`
So the `mayo-build` sub-folder is created alongside `mayo` and belongs to the same parent folder:
````bash
parent_folder
    ├── mayo
    └── mayo-build
````
`mayo-build` being the current directory then instruction `qmake ../mayo` is correct

closed 2023-09-04 by HuguesDelorme