fougue/mayo #218: Could not find Opencascade setting file?

easternbun, 2023-08-01:
program runs with error, it appeared at one runtime even when previous successful run was just seconds before and now every run has the error, how do you fix this?
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
[image]

HuguesDelorme, 2023-08-01:
Hello @easternbun
This is not an error but just a debug trace(this is not visible with release builds)
`opencascade.conf` is an optional file containing definitions of some OpenCascade environment variables(eg MMGT_OPT, MMGT_CLEAR, ...)
This file is used with the offical Windows version of Mayo.
If you want to silent this trace you can create an empty `opencascade.conf` file in the folder from where you start `mayo.exe`
But keep in mind that for your own Mayo debug builds you can safely ignore this warning.
Maybe this warning should be removed, thanks for your feedback

easternbun, 2023-08-02:
Thanks for replying @HuguesDelorme 
I placed a empty opencascade.conf under build and it does go through the checking file if exit but it throws a memory error upon further run.

HuguesDelorme, 2023-08-28:
Ah that was a bad idea then, just remove that empty opencascade.conf file and just ignore the corresponding DEBUG trace