fougue/mayo #219: a couple build and run time error

easternbun, 2023-08-02:
So I redownload the mayo and found a build error and a run time error
the build error:
GL_GLEXT_VERSION is redefined
The code line at 268 of command_system_information.cpp says QString cannot be converted to QChar
the run error:
memory error at run time when loading appModule settings

[image]
[image]
[image]
[image]

HuguesDelorme, 2023-08-02:
Hello @easternbun
Thanks for your feedback!

I've just pushed a fix in develop branch for the compiler error you reported(in command_system_information.cpp). This one happened for Qt6 and not Qt5. It reminds me to check compilation against Qt6 in the CI(ie GitHub Actions)

The one about GL_GLEXT_VERSION is a compiler warning(doesn't prevent compilation to succeed) and there's no fix I can find. This happens with OpenCascade 7.6.0 and with 7.7.0 version. You can ignore this warning, if it's really issue then try to update to OpenCascade 7.7.0

About the crash with Variant::toConstRefString() that one is weird. I suggest to clear the contents of the lastOpenDir setting in the registry and try again. Does the crash happens on the std::get<std::string>() or after with the creation of the FilePath object?
Would be great to see the callstack and paste here also what's inside the variant object as seen by Visual Studio(Code)

easternbun, 2023-08-03:
> 

@HuguesDelorme

Thanks for the reply! 
The crash happened at line 139 of settings.cpp "d->loadPropertyFrom" when trying to load lastOpenFolder property, and in the function loadPropertyFrom(), the program crashed at fromVariant(property, value). 
[image]
[image]

zerger, 2023-11-01:
i think maybe the path is contain chinese or  deleted

zerger, 2023-11-01:
[image]
[image]

zerger, 2023-11-01:
[image]

HuguesDelorme, 2023-11-02:
Hello @zerger
Thanks for the elaborated feedback
I've just pushed a fix for this issue on `develop` branch, can you please check if it's solved on your side?

HuguesDelorme, 2023-11-14:
Fixed with commit c2c1147

closed 2023-11-14 by HuguesDelorme