fougue/mayo #220: The Mayo application has encountered several performance issues on the Windows client, and in some cases, it's even incompatible after installation. This client is primarily intended for 3D file format previewing, including interactions with 3D files. WebAssembly (wasm), aside from being cross-platform and independent of the underlying operating system, offers excellent zero-copy functionality and performs well in file rendering.

long0419, 2023-08-19:
The Mayo application has encountered several performance issues on the Windows client, and in some cases, it's even incompatible after installation. This client is primarily intended for 3D file format previewing, including interactions with 3D files. WebAssembly (wasm), aside from being cross-platform and independent of the underlying operating system, offers excellent zero-copy functionality and performs well in file rendering.

@HuguesDelorme I can contribute to this project if possible. Rewriting it in Rust could also be a good option.

_Originally posted by @long0419 in https://github.com/fougue/mayo/issues/160#issuecomment-1684930200_

HuguesDelorme, 2023-08-19:
How wasm port of Mayo can be faster than current native app?
Can you please elaborate more on the performance issues of the Windows version of Mayo? And also the issues after installation? Maybe these problems can be fixed.
If it's about STEP file format there's nothing to be done about it as support is provided by OpenCascade 3rdparty library. Rust won't be of any help also for that.

long0419, 2023-08-20:
For the Windows version, it's not just about its current state. To make this app more successful, focusing on its extensibility is important. OpenCascade itself might not be ideal, perhaps limited to general format conversion. There are numerous ways to achieve compatibility and display STEP format. The strongest suit of this system lies in its approach to 3D file manipulation. I aim to transform this functionality into a web-based approach, avoiding the use of three.js.

Certainly, I can provide you with specific instructions on compiling this code locally.