fougue/mayo #223: cannot display .ply model exported from agisoft --win10

Wangpeiyu, 2023-09-01:
app version:mayo_v0.7.0.1113-f88a35d_winx64_installer.exe
os version:Windows 10 pro 22H2 19045.3324

cannot display( .ply, obj format ) 3d model and ply format point cloud data exported from agisoft.

But,i can see thubnail image of the loaded data in the welcome page.
[image]

nothing in main page:
[image]

sample data:
https://drive.google.com/file/d/1v9NyhWnLeJ9k85ILX8fq9GMfuWvC60d3/view?usp=drive_link

HuguesDelorme, 2023-09-01:
Hello @Wangpeiyu
The provided PLY model is far from the origin and as the origin trihedron graphics are taken into account for the "fit all in view" then the model is nearly invisible
Hide the origin trihedron graphics by unchecking menu _Display->Show Origin Trihedron_:
[image]
And then click on the "Fit All" button under the 3D view cube
See also issue #76

closed 2023-09-01 by HuguesDelorme

Wangpeiyu, 2023-09-08:
Thank you very much for your reply. It has solved a major problem for me.