fougue/mayo #224: Facing issue while running mayo ubuntu AppImage file

nikhil-convrseai, 2023-09-01:
WARNING: could not connect to display
INFO: Could not load the Qt platform plugin "xcb" in "" even though it was found.
FATAL: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)

Os: Ubuntu 22.04

I have followed the way of running the AppImage but still facing this issue.

HuguesDelorme, 2023-09-01:
Hello, can you please try to set environment variable QT_DEBUG_PLUGINS:
`$> export QT_DEBUG_PLUGINS=1`
Then start again Mayo and post here the ouput generated
Thanks!

nikhil-convrseai, 2023-09-01:
Hi this is the output

```
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo-0Blnq1i/usr/plugins/platforms" ...
DEBUG: QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Mayo-0Blnq1i/usr/plugins/platforms/libqxcb.so"
WARNING: Found metadata in lib /tmp/.mount_Mayo-0Blnq1i/usr/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

DEBUG: Got keys from plugin meta data ("xcb")
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo-0Blnq1i/usr/bin/platforms" ...
DEBUG: loaded library "/tmp/.mount_Mayo-0Blnq1i/usr/plugins/platforms/libqxcb.so"
WARNING: could not connect to display
INFO: Could not load the Qt platform plugin "xcb" in "" even though it was found.
FATAL: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Aborted (core dumped)```

HuguesDelorme, 2023-09-01:
Please check if the DISPLAY environment variable is set, if not try:
`$> export DISPLAY=:0`
And run Mayo again

nikhil-convrseai, 2023-09-01:
Hi I want to run Mayo on AWS EC2, where I don't have display, I want to use Mayo for converting a cad file to a 3d mesh file basically on my AWS instance.

HuguesDelorme, 2023-09-01:
This explains the issue then
Mayo GUI and CLI are delivered into a single executable so it requires display
However on Linux the xcb plugin is forced at startup within the `main()` function, see https://github.com/fougue/mayo/blob/1016ee9ad07fcc04575cecdb9816f8cfe900f2f0/src/app/main.cpp#L578

What is worth trying is the "offscreen" QPA plugin instead of "xcb"
Unfortunately to try this out you need to get and alter main.cpp and build Mayo locally

It's relevant to raise an improvement task for this, eg adding a command-line option to not force QPA plugin, this will let users the opportunity to test other plugins

nikhil-convrseai, 2023-09-01:
Hi 

while building locally, I followed the steps given in Readme file, but while executing following step

```qmake ../mayo CASCADE_INC_DIR=$CASCADE_INC_DIR CASCADE_LIB_DIR=$CASCADE_LIB_DIR```

I'm getting following error.

```Cannot find file: ../mayo.```

HuguesDelorme, 2023-09-04:
Hello please have a look at my answer to this issue #216
You are probably running qmake in the wrong directory

HuguesDelorme, 2024-08-07:
This is certainly fixed thanks to https://github.com/fougue/mayo/issues/260
@nikhil-convrseai feel free to open this issue again in case you experience the same troubles with coming 0.9.0 version

closed 2024-08-07 by HuguesDelorme