fougue/mayo #228: Undefined references when build

katyo, 2023-09-26:
I failed build v0.7.0 using qmake due to linking errors:
```
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.o: in function `Mayo::IO::OccFactoryReader::createProperties(Mayo::IO::Format, Mayo::PropertyGroup*) const':
io_occ.cpp:(.text+0x84): undefined reference to `Mayo::IO::OccObjReader::createProperties(Mayo::PropertyGroup*)'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.cpp:(.text+0xa4): undefined reference to `Mayo::IO::OccGltfReader::createProperties(Mayo::PropertyGroup*)'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.o: in function `Mayo::IO::OccFactoryWriter::createProperties(Mayo::IO::Format, Mayo::PropertyGroup*) const':
io_occ.cpp:(.text+0x141): undefined reference to `Mayo::IO::OccObjWriter::createProperties(Mayo::PropertyGroup*)'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.cpp:(.text+0x151): undefined reference to `Mayo::IO::OccGltfWriter::createProperties(Mayo::PropertyGroup*)'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.o: in function `Mayo::IO::OccFactoryReader::create(Mayo::IO::Format) const':
io_occ.cpp:(.text+0x1a9): undefined reference to `Mayo::IO::OccGltfReader::OccGltfReader()'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.cpp:(.text+0x301): undefined reference to `Mayo::IO::OccObjReader::OccObjReader()'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.o: in function `Mayo::IO::OccFactoryWriter::create(Mayo::IO::Format) const':
io_occ.cpp:(.text+0x530): undefined reference to `vtable for Mayo::IO::OccObjWriter'
/nix/store/zkjq96ik8cbv6ijh1lylnkk2bni9qvas-binutils-2.40/bin/ld: io_occ.cpp:(.text+0x606): undefined reference to `vtable for Mayo::IO::OccGltfWriter'
```

Qmake invoked as follows:
```
QMAKEPATH=/nix/store/6x5kri9biidq7j4v6q4bhbap4sbrci8l-qtbase-5.15.10-dev:/nix/store/xmlp78xr5lwqrwmcgm04p53h19fhfrr6-qttools-5.15.10-dev:/nix/store/qknlcpm7pw2wqcs9di94lz8bb53h5640-qtdeclarative-5.15.10-dev:/nix/store/51l6samhfcb63n9ssisqx08vm2qfg4ii-qtsvg-5.15.10-dev:/nix/store/zf18vjvk5k2sq1jn2jvsd2pidzpywymz-qtwayland-5.15.10-dev
qmake PREFIX=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0 NIX_OUTPUT_OUT=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0 NIX_OUTPUT_DEV=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0 NIX_OUTPUT_BIN=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0 NIX_OUTPUT_DOC=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0/share/doc/qt-5.15.10 NIX_OUTPUT_QML=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0/lib/qt-5.15.10/qml NIX_OUTPUT_PLUGIN=/nix/store/hwjmdk66c2i8b7c37wmyd076fa9n8i8r-mayo-0.7.0/lib/qt-5.15.10/plugins CONFIG+=release CONFIG+=nostrip QMAKE_CXXFLAGS=-I/nix/store/vvb5n0yg3j326y7bfwn7vaxy48pcykia-opencascade-occt-7.7.2/include/opencascade QMAKE_LRELEASE=/nix/store/xmlp78xr5lwqrwmcgm04p53h19fhfrr6-qttools-5.15.10-dev/bin/lrelease
```

The following command runs for linking:
```
g++ -Wl,-O1 -Wl,-rpath,/nix/store/wx2ckydd101ac55wbxw9nw21npnwqjsn-qtbase-5.15.10/lib -o mayo application.o application_item.o application_item_selection_model.o bnd_utils.o brep_utils.o caf_utils.o document.o document_tree_node.o document_tree_node_properties_provider.o enumeration.o geom_utils.o io_format.o io_reader.o io_system.o io_writer.o label_data.o math_utils.o mesh_access.o mesh_utils.o messenger.o messenger_client.o occ_progress_indicator.o occ_static_variables_rollback.o occt_enums.o point_cloud_data.o property.o property_builtins.o property_enumeration.o property_value_conversion.o quantity.o settings.o signal.o string_conv.o task.o task_manager.o task_progress.o text_id.o tkernel_utils.o triangulation_annex_data.o unit.o unit_system.o xcaf.o dxf.o io_dxf.o io_image.o io_occ.o io_occ_brep.o io_occ_caf.o io_occ_common.o io_occ_iges.o io_occ_step.o io_occ_stl.o io_occ_vrml_writer.o io_off_reader.o io_off_writer.o io_ply_reader.o io_ply_writer.o miniply.o ais_text.o graphics_create_driver.o graphics_create_virtual_window.o graphics_mesh_data_source.o graphics_mesh_object_driver.o graphics_object_driver.o graphics_point_cloud_object_driver.o graphics_scene.o graphics_shape_object_driver.o graphics_utils.o gui_application.o gui_document.o v3d_view_camera_animation.o v3d_view_controller.o measure_display.o measure_tool.o measure_tool_brep.o app_context.o app_module.o app_module_properties.o button_flat.o cli_export.o commands_api.o commands_display.o commands_file.o commands_help.o commands_tools.o commands_window.o console.o dialog_about.o dialog_inspect_xde.o dialog_options.o dialog_save_image_view.o dialog_task_manager.o document_property_group.o document_tree_node_properties_providers.o grid_helper.o gui_document_list_model.o item_view_buttons.o list_helper.o main.o mainwindow.o occt_window.o property_editor_factory.o property_item_delegate.o proxy_styled_item_delegate.o qsettings_storage.o qstring_utils.o qtgui_utils.o qtwidgets_utils.o recent_files.o theme.o widget_clip_planes.o widget_explode_assembly.o widget_file_system.o widget_gui_document.o widget_home_files.o widget_measure.o widget_message_indicator.o widget_model_tree.o widget_model_tree_builder.o widget_model_tree_builder_mesh.o widget_model_tree_builder_xde.o widget_occ_view.o widget_occ_view_controller.o widget_occ_view_impl.o widget_properties_editor.o format.o qrc_mayo.o moc_button_flat.o moc_commands_api.o moc_dialog_about.o moc_dialog_inspect_xde.o moc_dialog_options.o moc_dialog_save_image_view.o moc_dialog_task_manager.o moc_item_view_buttons.o moc_list_helper.o moc_mainwindow.o moc_property_item_delegate.o moc_widget_clip_planes.o moc_widget_explode_assembly.o moc_widget_file_system.o moc_widget_gui_document.o moc_widget_home_files.o moc_widget_measure.o moc_widget_message_indicator.o moc_widget_model_tree.o moc_widget_occ_view_controller.o   -L/nix/store/xfgkyzqz46jxdkrfjz2ksv9cyvaqcx8v-libGL-1.6.0/lib -lTKBin -lTKBinL -lTKBinXCAF -lTKBO -lTKBool -lTKBRep -lTKCAF -lTKCDF -lTKernel -lTKG2d -lTKG3d -lTKGeomAlgo -lTKGeomBase -lTKHLR -lTKLCAF -lTKMath -lTKMesh -lTKMeshVS -lTKOpenGl -lTKPrim -lTKService -lTKShHealing -lTKTopAlgo -lTKXSDRAW -lTKV3d -lTKVCAF -lTKXCAF -lTKXml -lTKXmlL -lTKXmlXCAF -lTKXSBase -lTKIGES -lTKXDEIGES -lTKSTEP -lTKSTEP209 -lTKSTEPAttr -lTKSTEPBase -lTKXDESTEP -lTKSTL -lTKVRML /nix/store/wx2ckydd101ac55wbxw9nw21npnwqjsn-qtbase-5.15.10/lib/libQt5Widgets.so /nix/store/wx2ckydd101ac55wbxw9nw21npnwqjsn-qtbase-5.15.10/lib/libQt5Gui.so /nix/store/wx2ckydd101ac55wbxw9nw21npnwqjsn-qtbase-5.15.10/lib/libQt5Core.so -L/nix/store/g1kflwfc5ym6xi59c16qscpr1ibhnyf1-libglvnd-1.6.0/lib -lGL -pthread
```

HuguesDelorme, 2023-09-26:
Hello @katyo 
What is the ouput of qmake command?
You should see something like:
````
Project MESSAGE: Mayo version 0.8.0.1183-61c473c debug
Project MESSAGE: Qt version 5.5.10
Project MESSAGE: OpenCascade version 7.7.2
Project MESSAGE: gmio OFF
````
The 3rd line is important because it proves Mayo qmake script identified correctly the OpenCascade version
Reason why I suspect there's an issue with OpenCascade is that you get linking errors on OBJ and gTF reader/writer and building or skipping the corresponding source files depends on the OpenCascade version

HuguesDelorme, 2024-01-11:
@katyo 
No activity, feel free to open again that issue if you still have those errors and more info to provide

closed 2024-01-11 by HuguesDelorme