fougue/mayo #229: How to solve the problem of importing step

thyis, 2023-10-09:
Dear sir, 
When I delete the document,   it done.
[image]
then select it to export,  it  done.
[image]
Finally, insert the exported document,  the graphics are not deleted but  position has changed,  how  to  solve.
[image]

thyis, 2023-10-11:
Hello, Dear Sir,
I have come to a solution that seemingly works: 
        TDF_Label shapeLabel = m_xcaf.shapeTool()->FindShape(m_xcaf.shapeTool()->GetShape(entityLabel));
	m_xcaf.shapeTool()->RemoveComponent(entityLabel);
	m_xcaf.shapeTool()->RemoveShape(shapeLabel);
        entityLabel.ForgetAllAttributes();
	entityLabel.Nullify();
	m_xcaf.shapeTool()->UpdateAssemblies();
Is this a valid solution? Do I need to do some cleanup to the document so it does not lead to some undefined behavior?

HuguesDelorme, 2023-10-11:
Thanks @thyis for your proposal
I will make some tests and merge the patch into the develop branch
At first sight I'm not sure if this is needed:
````c++
shapeLabel = shapeTool->FindShape(shapeTool->GetShape(entityLabel));
shapeTool->RemoveShape(shapeLabel);
````
Isn't the call to `shapeTool->RemoveComponent(entityLabel)` sufficient?

thyis, 2023-10-11:
Sometimes I just need to remove one of the components， ignore  "shapeTool->RemoveShape(shapeLabel)"，it it faild remove  shape.

closed 2023-10-16 by HuguesDelorme