fougue/mayo #232: about integrate mayo TO WPF

DDman1984, 2023-10-20:
Hi,
I plan to integrate mayo.exe into my WPF application. I've been trying to capture the progress of file conversion through the process (just like the text you showed in the example [](https://github.com/fougue/mayo/blob/develop/doc/screencast_cli.gif)), but the results have been frustrating...
How can I achieve the above requirement?

please give me any advice thank you very much:)

HuguesDelorme, 2023-10-20:
Hello @DDman1984
mayo.exe "refreshes" the whole contents of the progress text each time there's some update(eg task ended or progress change)
So if you connect to the standard ouput channel of mayo.exe this is what you should get with this pseudo-code:
````c++
while (mayoProcess.hasOutput) {
    string msg = mayoProcess.readAll();
    print(msg);
}
````

For example running `mayo.exe D:\dev\data\cax-if.org\as1-oc-214.stp --export D:\dev\data\cax-if.org\as1-oc-214.stp.stl` would produce these successive messages:
`0% Importing...`

`32% Importing...`

`37% Importing...`

...

`100% Imported`

````
100% Imported
0% Exporting as1-oc-214.stp.stl...
````

````
100% Imported
40% Exporting as1-oc-214.stp.stl...
````
...
````
100% Imported
100% Exporting as1-oc-214.stp.stl...
````

````
100% Imported
100% Exported as1-oc-214.stp.stl...
````

DDman1984, 2023-10-23:
Hi, @HuguesDelorme 

Thank you for your advice and response. In fact, I've tried using C# process APIs and consulted the .NET official documentation. However, I still haven't been able to obtain Mayo's output log during the conversion process.

I'm including my code below in the hope that it can help you understand my issue. Thank you very much.

`namespace WpfApp1
{
    /// <summary>
    /// Interaction logic for MainWindow.xaml
    /// </summary>
    public partial class MainWindow : Window
    {
        public MainWindow()
        {
            this.InitializeComponent();

            var proc = new Process
            {
                StartInfo = new ProcessStartInfo
                {
                    FileName = "C:\\Program Files\\Fougue\\Mayo\\mayo.exe",
                    Arguments = "C:\\Users\\User\\Downloads\\QG76_CFM_5p.stp --export C:\\Users\\User\\Downloads\\test.glb",
                    UseShellExecute = false,
                    RedirectStandardOutput = true,
                    CreateNoWindow = false
                }
            };

            proc.Start();

            string q = "";
            while (!proc.HasExited)
            {
                q += proc.StandardOutput.ReadToEnd();
            }
            Console.WriteLine($"output : {q} ");

        }
    }
}`

The above code can launch Mayo and perfectly convert the model to the glb format, but I'm unable to retrieve any output text from the C# process...@@