fougue/mayo #244: Portable version

git-tiger, 2023-12-15:
Is there option to run the viewer without installing. My company does not allow installation of software.

HuguesDelorme, 2023-12-15:
Work has to be done for a WebAssembly port. So it will be possible to safely run Mayo with eg wasmer

HuguesDelorme, 2023-12-18:
@git-tiger 
Do you mean having instead a ZIP archive of the binaries you can extract somewhere and directly run `mayo.exe`?

git-tiger, 2023-12-18:
Yes, no need to install.

HuguesDelorme, 2023-12-18:
Version 0.8.0 will be shipped this week, I will try to provide a ready-to-use ZIP archive containing the binaries

closed 2023-12-20 by HuguesDelorme

HuguesDelorme, 2023-12-20:
Choose `Mayo-0.8.0-win64-binaries.zip` in the [Releases](https://github.com/fougue/mayo/releases/tag/v0.8.0) page

git-tiger, 2023-12-21:
Unfortunately, it flagged as Reputation virus, and quarantined the mayo.exe file.
Office computer is using Symantec Antivirus.

HuguesDelorme, 2023-12-21:
Mayo binaries were analyzed by up-to-date avast anti-virus and no infection was reported.
Might probably be a Symantec false-positive and right now there's nothing I can do to workaround this.
In the future I will sign Mayo binaries with an official certificate, that should solve the issue.