fougue/mayo #250: About warning of import dxf

thyis, 2023-12-28:
Hello dear developers,
I imported a dxf model I designed. As you can see from the image below, view are none.
[image]
[dxf.zip](https://github.com/fougue/mayo/files/13782317/dxf.zip)

HuguesDelorme, 2024-01-09:
After investigation it appears that some BSpline entities are not correct for OpenCascade.
For example first BSpline entity in `88732000.dxf`:
- degree is 3
- it's non-periodic(flags = 8)
- 16 control points(poles)
- 20 knots(5 unique knots)
- Multiplicities(mults) of the unique knots are: 4, 4, 4, 4, 4

OpenCascade states that each multiplicity must be <= `degree`(or `degree+1` at the first and last knot of a non-periodic curve).
For that entity, BSpline mults are incorrect.
Which software did you use to generate these files?

thyis, 2024-01-11:
ZWsoft,  A very popular design software in China in recent years

HuguesDelorme, 2024-01-18:
As you are a commercial user of ZWsoft that would be great if you ask them about this issue. The software is creating BSPKINE entities that seem to be ill-formed(at least for OpenCascade)
Come back here if you have feedback from ZWsoft dev

Mannshoch, 2025-01-30:
I compared an Import from this [DXF sample file](https://gitlab.com/inkscape/extensions/-/merge_requests/629#note_2031419407) with FreeCAD. In FreeCAd I had a better result. But FreeCAD asks me about using either native or SAT files. Here the SAT data had a better results than native.