fougue/mayo #254: IGS File is breaking on debian build

nikhil-convrseai, 2024-02-02:
[image]

Hi, when i'm converting iges file from debian build to glb it is breaking it like this whereas when i'm trying it in macos build it is working fine.

on ubuntu i'm doing it via command line on a server in aws.

here is the view of file on macos the original and correct file:

[image]

nikhil-convrseai, 2024-02-02:
also when i'm using step of the same file

it is removing the material

[image]

whereas original is perfectly fine

[image]

HuguesDelorme, 2024-02-05:
Hello can you attach the igs and step files so I can try to reproduce the issues?

nikhil-convrseai, 2024-02-05:
Hi

Please find the attached files
 Montratec Log.igs
<https://drive.google.com/file/d/1NFc8hirZ2rlgZuov4kerbRu2dAuIMksH/view?usp=drive_web>
 MontratecLogistik.step
<https://drive.google.com/file/d/1CYaOv29jPVTnpjuCYbUK0t6fdUWFsost/view?usp=drive_web>

HuguesDelorme, 2024-02-06:
Yes and I think the colors(materials) seem correct in the step->gltf case. It's just that the gltf rendering in Mayo has an issue with lighting(too much intensity)

nikhil-convrseai, 2024-02-07:
hi @HuguesDelorme were you able to find the issue with IGES? it is a bit critical for us to solve this.

HuguesDelorme, 2024-02-07:
I'm getting different imports(for the IGES file) when Mayo is built with OpenCascade 7.6 and 7.8
With OpenCascade 7.6:
[image]

With OpenCascade 7.8:
[image]

When trying to load the gltf export I'm getting a crash in Mayo
Is that you'r experiencing as well?

nikhil-convrseai, 2024-02-08:
no my issue is that when on ubuntu i pass the igs file through mayo it breaks the file into small pieces it is not coming inact like you are getting. 

I'm using ubuntu 22.04, which has opencascade 7.5 can that be the issue?

nikhil-convrseai, 2024-02-08:
Hi @HuguesDelorme 
 I upadted my Opencascade to 7.6 on ubuntu and was built mayo with the same and now i'm also facing same issue of export i.e. it is crashing while exporting gltf. how can i solve that ?

HuguesDelorme, 2024-02-08:
After investigation it happens that there's a data race in Mayo when creating a model tree(just after file import) that contains many root entities. This is the case for this IGES file
I'm currently implementing a fix, will let you know when it's ready in develop branch

HuguesDelorme, 2024-02-08:
@nikhil-convrseai
The crash should be fixed with latest changes in `develop` branch
Let me know if it works also on your side

nikhil-convrseai, 2024-02-09:
@HuguesDelorme I tried but it still failing while exporting glb,

my environment is ubuntu 22.04 and opencascade version 7.6.0

HuguesDelorme, 2024-02-09:
I've tested in virtual machine running ubuntu 20.04 and Mayo was built in debug with OpenCascade 7.7.0
Exporting the "Montratec Log.igs" file to glTF works fine:

[image]

nikhil-convrseai, 2024-02-12:
@HuguesDelorme I'm using mayo with CMD only and I'm getting 

```DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
 32% Importing...

Error during export to '/home/ubuntu/convrse/meshes/high/QEGozO0Y6EgZa7Jy0uoS9/TK_ymQDis5hw/high/OJZCJpwgl0Vj/montratec_log.glb' File
 write problem
100% Imported```

error when exporting gltf now opencascade version is 7.7.0 only and ubuntu is 22.04

HuguesDelorme, 2024-02-13:
You need to install RapidJSON library for glTF output with OpenCascascade
Install it with eg `sudo apt-get install rapidjson-dev` and run cmake step for OpenCascade.
You probably need to rebuild OpenCascade again, not sure

nikhil-convrseai, 2024-02-14:
hi @HuguesDelorme thanks for the continuous support it is working at my end as well, let me know when you merge `develop` branch into `main`

HuguesDelorme, 2024-02-14:
Good to know it works for you!
`develop` branch is merged into `master` only on new release(it will happen next time when releasing 0.9.0 version)

closed 2024-02-14 by HuguesDelorme