fougue/mayo #257: Build should use pkg-config (or CMake?)

mmuman, 2024-02-19:
I noticed the Linux build instructions use weird `dpkg` incantations what what should really just be calls to `pkg-config`. Using `dpkg` means people not using Debian-based distros would need to figure out themselves.

While `qmake` supports using `pkg-config` to find packages, it doesn't seem to support optional dependencies though, so it errors out if it doesn't find `libassimp`… Also, OpenCASCADE doesn't seem to install any `.pc` file so it wouldn't be discoverable by `pkg-config`.

Another option would be to switch to CMake for configuration I suppose…

mmuman, 2024-02-19:
Too lazy to create an account on the OCCT bug tracker to tell them about `pkg-config` right now…

HuguesDelorme, 2024-02-19:
Hello @mmuman
Yes CMake support is currenty work-in-progress in branch [port/cmake](https://github.com/fougue/mayo/tree/port/cmake)
It will be merged into `develop` branch soon(hopefully this week)

mmuman, 2024-02-19:
Trying it here it fails to build:
```
$ cmake ..
-- The CXX compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- Qt version 5.15.10
-- OpenCascade version 7.6.3
CMake Warning at CMakeLists.txt:414 (message):
  VRML reader disabled because OpenCascade < v7.7

-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/revol/devel/mayo/mayo/build
$ make -j2
[  4%] Building CXX object CMakeFiles/mayo.dir/mayo_autogen/mocs_compilation.cpp.o
In file included from /home/revol/devel/mayo/mayo/src/app/../base/document.h:9,
                 from /home/revol/devel/mayo/mayo/src/app/commands_api.h:9,
                 from /home/revol/devel/mayo/mayo/src/app/app_context.h:9,
                 from /home/revol/devel/mayo/mayo/src/app/app_context.cpp:7:
/home/revol/devel/mayo/mayo/src/app/../base/application_ptr.h:9:10: fatal error: TDocStd_Application.hxx: Aucun fichier ou dossier de ce type
    9 | #include <TDocStd_Application.hxx>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Could it be it forgets to add the include dir to the list for Linux?

HuguesDelorme, 2024-02-19:
Yes I'm having the same issue for Linux CI
OpenCascade cmake package files started to add INTERFACE_INCLUDE_DIRECTORIES starting from 7.7.0 version
Older versions require to explicitly add OpenCascade include dir
Will have to workaround this

mmuman, 2024-02-19:
Just tested again, it builds :-)

HuguesDelorme, 2024-02-20:
Nice to see it works on your side! CMake build works fine on Ubuntu 20.04 as well(tested in dedicated VM)
But I'm having hard times to make it work on Linux CI, getting this weird error at the end of build:
`gmake[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libtbb.so', needed by 'mayo'.  Stop.`
Any idea what could cause this?

HuguesDelorme, 2024-02-20:
Linux CI(Github workflow) runs on Ubuntu 22.04 and provides OpenCascade 7.5.1
Looking at package file `/usr/lib/cmake/opencascade/OpenCASCADEFoundationClassesTargets.cmake` shows that it wants to link with `/usr/lib/x86_64-linux-gnu/libtbb.so`:
````cmake
# Create imported target TKernel
add_library(TKernel SHARED IMPORTED)

set_target_properties(TKernel PROPERTIES
  INTERFACE_LINK_LIBRARIES "pthread;rt;stdc++;/usr/lib/x86_64-linux-gnu/libtbb.so;/usr/lib/x86_64-linux-gnu/libtbbmalloc.so;dl"
)
````
But libtbb2 package was installed as a dependency and provides `/usr/lib/x86_64-linux-gnu/libtbb.so.2`(not `libtbb.so`)

mmuman, 2024-02-20:
Yay!

HuguesDelorme, 2024-02-22:
Will be addressed with #53

closed 2024-02-22 by HuguesDelorme