fougue/mayo #260: Only cli usage

Spoutnik97, 2024-03-01:
Hi, thanks for your work, and the documentation!
Do you have advices to make a light version of your app without GUI ? I would like to use it in a docker container, so the lightest version is the best

HuguesDelorme, 2024-03-06:
So you would like to only use the converter from command-line?
Ideally without any Qt dependency?

Spoutnik97, 2024-03-13:
Yes exactly

I would like to have the lightest docker image to convert files from an api

HuguesDelorme, 2024-03-22:
Hello @Spoutnik97
I've made Mayo GUI and CLI distinct application in develop branch
See new respective CMake options `Mayo_BuildApp` and `Mayo_BuildConvCli`(both ON by default)
The batch converter is now a distinct exe(`mayo-conv.exe` on Windows) and it links to the same OpenCascade runtime DLLs as the GUI app, but for Qt only QtCore is linked to(no QtGui, QtWidgets, QtOpenGL and QtSvg)

closed 2024-03-25 by HuguesDelorme