fougue/mayo #261: Segmentation Fault Image Export

ParthivS20, 2024-03-25:
Hello, I am trying to export a STEP file to an image file and receive a segmentation fault. The exported image file opens properly, but still shows a segmentation fault at the end of execution. I am using the compiled AppImage on Ubuntu 22.04.3.

./mayo.AppImage -s mayo-config.ini input.step -e output.png

DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
 40% Exporting test.png...
100% Imported
Segmentation fault

I tried using different 3d formats and image formats but still got the same error. Converting to different 3D formats works.

HuguesDelorme, 2024-03-25:
Hello @ParthivS20
Is this an AppImage you built yourself? The AppImage provided in Releases page was made on Ubuntu 20.04(latest LTS still supported today)
I will try to reproduce the issue soon, thanks for reporting that issue

ParthivS20, 2024-03-25:
It’s the pre-compiled AppImage in release version 0.8.0. Thanks for looking into it!

HuguesDelorme, 2024-03-26:
@ParthivS20 
I couldn't reproduce the crash
Tested on Ubuntu 20.04 running in dedicated VM with VirtualBox
Can you share the settings file `mayo-config.ini` you are using?
And possibly also the input STEP file

ParthivS20, 2024-03-26:
Sure. I am running on Ubuntu 22.04 if that could cause an issue. Here is the config file

```
[meshing]
meshingQuality=VeryPrecise

[import]
STEP\assemblyLevel=Structure
STEP\encoding=UTF8
STEP\preferredShapeRepresentation=AdvancedBRep
STEP\productContext=Design
STEP\readShapeAspect=true
STEP\readSubShapesNames=false

[export]
Image\backgroundColor=#A4A8B0
Image\cameraOrientation="1, -1, 1"
Image\cameraProjection=Orthographic
Image\height=1080
Image\width=1920
```

ParthivS20, 2024-03-26:
I did some further testing. Segmentation Fault occurs when running in WSL, but the exports complete successfully. Ran the same config and STEP files in 20.04 and 22.04 VMs. The AppImage worked fine on both, just needed to install libfuse2 in 22.04. I ran the AppImage in GitHub action runners on 20.04 and 22.04 and the conversions seem to fall into an infinite loop when exporting to an image (STEP->STL conversion works). Attached the log below.

```
Run ./mayo.AppImage docs/DCDC-FusePCB.step -e docs/DCDC-FusePCB.stl
  ./mayo.AppImage docs/DCDC-FusePCB.step -e docs/DCDC-FusePCB.stl
  ./mayo.AppImage docs/DCDC-FusePCB.step -e docs/DCDC-FusePCB.png
  exit 0
  shell: /usr/bin/bash -e {0}
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
  0% Importing...

 3[2](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:2)% Importing...

 [3](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:3)3% Importing...

 3[4](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:4)% Importing...

 36% Importing...

 37% Importing...

 38% Importing...

 39% Importing...

 [5](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:5)1% Importing...

 75% Importing...

 77% Importing...

 79% Importing...

100% Imported    

100% Imported

  0% Exporting DCDC-FusePCB.stl...
100% Imported

 40% Exporting DCDC-FusePCB.stl...
100% Imported

 42% Exporting DCDC-FusePCB.stl...
100% Imported

 51% Exporting DCDC-FusePCB.stl...
100% Imported

 51% Exporting DCDC-FusePCB.stl...
100% Imported

 51% Exporting DCDC-FusePCB.stl...
100% Imported

 54% Exporting DCDC-FusePCB.stl...
100% Imported

 57% Exporting DCDC-FusePCB.stl...
100% Imported

 [6](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:7)0% Exporting DCDC-FusePCB.stl...
100% Imported

 63% Exporting DCDC-FusePCB.stl...
100% Imported

 66% Exporting DCDC-FusePCB.stl...
100% Imported

 69% Exporting DCDC-FusePCB.stl...
100% Imported

 [7](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:8)2% Exporting DCDC-FusePCB.stl...
100% Imported

 75% Exporting DCDC-FusePCB.stl...
100% Imported

 78% Exporting DCDC-FusePCB.stl...
100% Imported

 81% Exporting DCDC-FusePCB.stl...
100% Imported

 84% Exporting DCDC-FusePCB.stl...
100% Imported

 87% Exporting DCDC-FusePCB.stl...
100% Imported

 [9](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:10)0% Exporting DCDC-FusePCB.stl...
[10](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:11)0% Imported

 93% Exporting DCDC-FusePCB.stl...
100% Imported

 96% Exporting DCDC-FusePCB.stl...
100% Imported

 99% Exporting DCDC-FusePCB.stl...
100% Imported

100% Exporting DCDC-FusePCB.stl...
100% Imported

100% Exported DCDC-FusePCB.stl    
100% Imported
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
  0% Importing...

 [40](https://github.com/lhr-solar/DCDC-FusePCB/actions/runs/8442894911/job/23125325768#step:4:41)% Importing...

 41% Importing...

 42% Importing...

 43% Importing...

 44% Importing...

 45% Importing...

 46% Importing...

 48% Importing...

 51% Importing...

 52% Importing...

 53% Importing...

 55% Importing...

 56% Importing...

 58% Importing...

 73% Importing...

100% Importing...

100% Imported    

  0% Exporting DCDC-FusePCB.png...
100% Imported

 40% Exporting DCDC-FusePCB.png...
100% Imported

100% Exporting DCDC-FusePCB.png...
100% Imported
```

HuguesDelorme, 2024-04-11:
When you run `Mayo.AppImage` within GitHub action workflow you may need to start X11 display server before conversion:
````
        - name: Convert STEP to image
          env:
              DISPLAY: :0
          run: |
              Xvfb $DISPLAY -screen 0 1280x1024x24 &
              sleep 5s # Maybe not needed
              Mayo.AppImage docs/DCDC-FusePCB.step -e docs/DCDC-FusePCB.png
              
````
Let me know if that helps

ParthivS20, 2024-04-11:
That worked, thank you! I was previously running it with Xvfb but seems like I was missing the sleep timer.

closed 2024-04-11 by ParthivS20