fougue/mayo #262: Export each part as separate STL and STEP file from console

semixor, 2024-03-25:
Hello,
is it possible to export each part of CAD (STEP) file as a separate STL and also STEP file from console ?

HuguesDelorme, 2024-04-04:
Hello @semixor
For your purpose I think the most flexible solution would be scripting support
This has to be implemented and then scripting in Mayo would allow you to visit assemblies by JS(JavaScript) code and do whatever is required(eg export parts to STL files in your case)

Mayo should also support executing a scripting file from command line and exit

What do you think of that solution?

semixor, 2024-04-04:
Hello @HuguesDelorme,
Absolutely ! This is the best solution !
Also would be nice to have some control over it, like ability to export the wires(edges) for later visualization in ThreeJS for example etc.

Looking forward when you will be able to implement it, many thanks, can't wait !

semixor, 2024-05-02:
Hello @HuguesDelorme , any status update on this ?

HuguesDelorme, 2024-05-02:
Hello @semixor
I started to work on scripting support in the [`feature/scripting`](https://github.com/fougue/mayo/tree/feature/scripting) branch
For now it's providing basic DOM-like access to Mayo data, here is an testing example script:
````javascript
console.debug("Mayo version: " + application.versionString);
console.debug("Document count: " + application.documentCount);
for (let i = 0; i < application.documentCount; i++) {
    let doc = application.documentAt(i); // Access i-th document
    console.debug("Document: " + doc.name);
    console.debug("    filePath: " + doc.filePath);
    console.debug("    entityCount: " + doc.entityCount);
}

if (application.documentCount > 0) {
    let doc = application.documentAt(0); // Access 1st document
    doc.traverseModelTree(
        nodeId => {
            console.debug(
                "treeNodeId: " + nodeId
                + " name: " + doc.treeNodeName(nodeId)
                + " parentId: " + doc.treeNodeParent(nodeId)
                + " tag: " + doc.treeNodeTag(nodeId)
            );
        }
    );
}
````

The next important things to be supported are:
- access to BRep shapes and their underlying geometry
- reader/writer per file format

For writer objects it's not easy to support what you ask "ability to export faces of a part and wires(edges)" because the smallest thing that can be exported is a "node" in the mode tree. A model tree node in Mayo being an assembly, instance or part.
Maybe the solution is to make writer objects support exporting of `TDF_Label` directly. So if you want to export a shape(face, wire, ...) then the scripting API should allow to add a label for that shape(XDE) and pass the label to the writer.

semixor, 2024-05-06:
Hi @HuguesDelorme this is very nice, looking forward to be able to work with it, great job, thank you !

semixor, 2024-05-14:
Hi @HuguesDelorme when I will be able to test this beautiful work :) ? Thanks !

HuguesDelorme, 2024-05-17:
@semixor
I've been working on different topics on Mayo these last weeks, I should have focused on a single one...
Going to work on JS scripting again, looks like a very valuable feature for the user
Will let you know about significant work pushed on the `feature/scripting` branch

semixor, 2024-05-17:
Hi @HuguesDelorme ,
Thank you so much for the update !

semixor, 2024-06-06:
Hi @HuguesDelorme,
I see there is some active development happening for this feature, that's awesome !
Please let me know when you have first working version to test, thank you so much !
Best Regards !

HuguesDelorme, 2024-06-06:
@semixor
Access to BRep shapes are now supported by scripting
There's almost 100% support for surface geometry access(plane, cylinder, bspline, ...)
Next step is to cover also curve geometry from BRep edge
The step after is supporting reader/writer objects

semixor, 2024-06-06:
@HuguesDelorme wow that's a lot of work being done from your side, thank you.

IvanLloro, 2024-10-26:
This scripting feature is great.

The only thing I am missing to cover my personal need is a way of exporting the text output of the script either by allowing creating a TXT file in the filesystem (preferred) or by allowing copying the Javascript console content so it can be manually pasted somewhere else.