fougue/mayo #263: Error during export to '*.gltf' File write problem

hjxy2012, 2024-03-28:
I'm using mayo-conv.exe on windows 10. It was built using the latest "develop" branch code with msvc 2019 + qt 5.15.2 + occ 7.8.0. 
After I typed "mayo-conv *.step --export *.gltf"，I got "Error during export to *.gltf File write problem". My mayo-conv.exe worked well when I tried to convert the step to iges/obj/stl. And when I tried to open the step with mayo GUI and export the geometry to .gltf, it worked well. So what's with my mayo-conv.exe?

HuguesDelorme, 2024-03-28:
Hello @hjxy2012

`mayo-conv` doesn't work with file globbing as you tried
You should use instead a shell forloop:
`for %f in ("D:\data\*.stp") do (mayo-conv.exe --no-progress "%f" --export "%~dpnf.gltf")`

This will convert STEP files contained in `D:\data` folder to glTF format, keeping the same base name.

hjxy2012, 2024-03-28:
> You should use instead a shell forloop:

Thank you for your reply. I tried to run following your instructions but it still didn't work. Maybe I should debug it and get the answer. And the error as follows: CRITICAL: "Error during export to '*.gltf'\nFile write problem "

HuguesDelorme, 2024-03-28:
From the CRITICAL message, it seems you are still using file globbing(*.gltf)

hjxy2012, 2024-03-29:
> From the CRITICAL message, it seems you are still using file globbing(*.gltf)

It seems m_seqRootLabel of io_occ_gltf_writer is empty in my case. I'm still trying to solve this by debugging.

HuguesDelorme, 2024-03-29:
If you built OpenCascade yourself, make sure you enabled RapidJson library as it is required by glTF toolkit.
When RapidJson is off it doesn't prevent OpenCascade to build, but it causes glTF reader/writer to fail at runtime

hjxy2012, 2024-03-29:
> If you built OpenCascade yourself, make sure you enabled RapidJson library as it is required by glTF toolkit. When RapidJson is off it doesn't prevent OpenCascade to build, but it causes glTF reader/writer to fail at runtime

I build OpenCascade 7.7.2 with option  "SET USE_RAPIDJSON ON" and rebuild mayo.  Converting step to gltf finally worked! Thank you.  

By the way, does mayo and mayo-conv use different method to export gltf file？Mayo.exe worked well when there is no Rapidjson in OpenCascade, but mayo-conv.exe went wrong.