fougue/mayo #274: CLI not working properly

nikhil-convrseai, 2024-04-18:
Hi 

I have been using mayo as cli for past many months, but recently i have started facing the following error when running it in cli.

```
./mayo /home/ubuntu/convrse/meshes/high/ZPVqGx23IMtU_JkK4HnfO/TK_ymQDis5hw/high/SLhmmUxTWZgc/valvula_10_600.step --export x.glb
WARNING: could not connect to display
INFO: Could not load the Qt platform plugin "xcb" in "" even though it was found.
FATAL: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted
```

if i'm running cli, it should not expect display i guess, please help as it has caused issue in of our major production pipeline.

HuguesDelorme, 2024-04-19:
Hello
If you built Mayo from source code yourself then the converter CLI is in a distinct executable named `mayo-conv` in `develop` branch(see #260)
This executable only needs QtCore from Qt framework and should not require a Qt platform plugin to run.
Notice if you want to convert CAD files to images(PNG, JPEG, ...) you will need to have `Xvfb` running(see #261)

HuguesDelorme, 2024-05-06:
@nikhil-convrseai
Are you still having the issue?

HuguesDelorme, 2024-05-17:
@nikhil-convrseai 
Any update on your side about the reported issue?

nikhil-convrseai, 2024-05-17:
Hi @HuguesDelorme it is working fine now thanks.

closed 2024-05-20 by HuguesDelorme