fougue/mayo #276: mayo vs mayo-conv different export

semixor, 2024-05-07:
Hi @HuguesDelorme,
the issue is following:
1. When converting STEP to VRML from GUI the wrl file contains lines (edges)
2. When converting STEP to VRML from MAYO-CONV (CLI) the wrl file doesn't contain lines (edges)

what can cause the issue ? is there any paramater I need to set ?

HuguesDelorme, 2024-05-13:
Mayo(GUI app) and mayo-conv use distinct settings
Try to force mayo-conv use the same settings as Mayo GUI:
1. Open Mayo GUI application
2. Click on _Tools->Options_ menu
3. Click _Exchange->Save as..._ button(in the Options dialog)
4. Select an INI file(assume `D:\data\mayo-gui.ini` in this procedure) and confirm file dialog
5. From command prompt run:
````bash
$> mayo-conv --use-settings D:\data\mayo-gui.ini inputfile.step --export outputfile.wrl
````

semixor, 2024-05-14:
@HuguesDelorme unfortunately this is not working, I did it from the beginning to be honest and still the result is same as I mentioned

HuguesDelorme, 2024-05-17:
@semixor
I couldn't reproduce the problem with non-trivial step files, both mayo and mayo-conv produce the "same" file regarding edges
Can you please provide the input STEP file?
Also please give here your mayo-conv settings cache:
`$> mayo-conv --write-settings-cache semixor-mayoconv.ini`

HuguesDelorme, 2024-08-02:
@semixor
Any news on that issue?

Kahlis, 2024-09-25:
@HuguesDelorme 
I'm facing the same issue, should try your solution soon and give feedback here asap