fougue/mayo #279: Suggested enhancement to "reload documents on file changes" option

mortisj, 2024-06-24:
Mayo has a hot reload option that causes a popup window to display if the file being displayed has changed.  I wonder if that checkbox could be turned into a 3-way option:
- don't monitor open files for changes
- display a dialog to ask if changed file should be reloaded
- automatically reload changed files with no user interaction

The 3rd option would save a bit of clicking while developing models using a tool that does not have a built-in GUI (build123d in my case).

Thanks.

eddieliberato, 2024-07-29:
Hey,

I would like to upvote this !!! I also would like to see an option to enable the hot reload automatically, without the need to click to accept the reload !

---
I will leave my temporary hack here, maybe it will benefit someone with the same needs

on the file /src/app/widget_main_control.cpp (around line 330) I modify the original method with the following:

```cpp
#include <QTimer>

void WidgetMainControl::reloadDocumentAfterChange(const DocumentPtr& doc)
{
    const QString strQuestion =
        tr("Document file `%1` has been changed since it was opened\n\n"
           "Do you want to reload that document?\n\n"
           "File: `%2`")
        .arg(to_QString(doc->name()))
        .arg(QDir::toNativeSeparators(filepathTo<QString>(doc->filePath())));
    const auto msgBtns = QMessageBox::Yes | QMessageBox::No;
    auto msgBox = new QMessageBox(QMessageBox::Question, tr("Question"), strQuestion, msgBtns, this);
    msgBox->setTextFormat(Qt::MarkdownText);
    QtWidgetsUtils::asyncDialogExec(msgBox);

    // Set up a timer to automatically click the Yes button after a short delay
    QTimer::singleShot(10, this, [msgBox]() {
        msgBox->button(QMessageBox::Yes)->click();
    });

    QObject::connect(msgBox, &QMessageBox::buttonClicked, this, [=](QAbstractButton* btn) {
        m_docFilesWatcher->acknowledgeDocumentFileChange(doc);
        if (btn == msgBox->button(QMessageBox::Yes)) {
            while (doc->entityCount() > 0)
                doc->destroyEntity(doc->entityTreeNodeId(0));

            FileCommandTools::importInDocument(m_appContext, doc, doc->filePath());
        }
    });
}
```

this is basically a very short timer to accept the changes automatically. When I tried to remove invocation of the message box completely I got an error. Seems like there is a callback waiting for the message box, idk.

---

I like mayo a lot, this would be awesome! 

All the best !

HuguesDelorme, 2024-07-30:
Option is now an enumeration to choose the action after a file change:
- None
- Reload If User Confirms
- Reload Silently

[image]

closed 2024-07-30 by HuguesDelorme