fougue/mayo #281: QOpenGLWidget displays black screen

FrankZNL, 2024-07-26:
[image]
1. Under Windows system, the opencascade version is 7.7, use a laptop to connect the external screen, and set the external screen as the main screen
2. Turn off the external screen or unplug the HDMI cable;
3. The software automatically switches to the laptop and the software displays
black screen

HuguesDelorme, 2024-07-26:
Can you please post what's reported by _Help->System Information..._ dialog?
Use the _Copy to Clipboard_ button

Seems like a minor bug to me, what's the behavior if you set back the main screen to laptop with Windows(instead of unplugging HDMI cable)?

FrankZNL, 2024-07-26:
Thank you!
[image]
If I use win+P to set it to the main screen to laptop only, a black screen will also appear. 

Looking forward to your reply!

FrankZNL, 2024-07-26:
I captured the event using QAbstractNativeEventFilter, but I haven’t found a suitable way to reconnect the context later. Thanks!
bool nativeEventFilter(const QByteArray& event_type, void* message, long* result)
{
#ifdef Q_OS_WIN
    if (event_type == "windows_generic_MSG")
    {
        MSG* msg = reinterpret_cast<MSG*>(message);
        if (msg->message == WM_DISPLAYCHANGE)
        {
            QMetaObject::invokeMethod(this, "SlotDisplayChanged", Qt::QueuedConnection);
        }
    }
#endif
}

FrankZNL, 2024-07-29:
I fixed the issue by recreating a new Aspect_NeutralWindow(). Thanks!

Mannshoch, 2025-01-27:
Have the same problem.  On Friday I used Mayo to show a file. Then left the PC. Now I came back and the screen is black.

It seems only a temporary problem. I could open other files that works flawless beside this file and if I close it and reopen it also appear correct again. I assume that either standby mode or another influence has caused the problem.