fougue/mayo #282: some model types in dxf data cannot be read

NoWhere-NowHere-TLD, 2024-07-28:
e.g the four surface types in dxf(ExtrudedSurface,SweptSurface...).
Maybe it doesn't support 3dsolid.
Do you plan to support these types?
thanks for you looking at this question

HuguesDelorme, 2024-07-29:
Mayo doesn't support those types yet.
Current development version supports 3D meshes(see this [issue](https://github.com/fougue/mayo/issues/251))
Can you provide some sample DXF files for these surface types?
It's probably easy to support them with OpenCascade

NoWhere-NowHere-TLD, 2024-07-29:
thanks for your answer.
How can I give you the SweptSurface dxf file, I think it might not be appropriate to put it directly into the github project.
If you don't mind I can put it.
And I find the group codes 310 in the dxf file, but when I convert the binary codes to ascii myself, I got messy code.
I'm the newbie to using OpenCascade(also parsing of dxf), could it parsing the dxf? I am planning to use dxflib or [libdxf](https://github.com/bert/libdxf), it could process the soft or hard pointers? I feel a little daze about it

NoWhere-NowHere-TLD, 2024-07-29:
> Mayo doesn't support those types yet. Current development version supports 3D meshes(see this [issue](https://github.com/fougue/mayo/issues/251)) Can you provide some sample DXF files for these surface types? It's probably easy to support them with OpenCascade

thanks for your answer.
How can I give you the SweptSurface dxf file, I think it might not be appropriate to put it directly into the github project.
If you don't mind I can put it.
And I find the group codes 310 in the dxf file, but when I convert the binary codes to ascii myself, I got messy code.
I'm the newbie to using OpenCascade(also parsing of dxf), could it parsing the dxf? I am planning to use dxflib or [libdxf](https://github.com/bert/libdxf), it could process the soft or hard pointers? I feel a little daze about it

HuguesDelorme, 2024-07-30:
Hello you can try to put the DXF files in a ZIP archive and share it here in this thread
OpenCascade provides support of DXF format but only through a commercial and closed component

NoWhere-NowHere-TLD, 2024-07-30:
[dxfSample.zip](https://github.com/user-attachments/files/16425418/dxfSample.zip)
Thank you for your reply.
The purpose of parsing dxf is to convert it to obj, It's a part of my supervisor's assignment, and he can't help me much more on this.
I couldn't find an open source library to do this, so I try to write(add) it myself.
I looked at the code for reading dxf in mayo, but I couldn't find a place to start.
I am now looking for the relevant information of dxf to see(like [sat format](https://paulbourke.net/dataformats/sat/sat.pdf), [dxf R13 reference](https://filemonger.com/specs/dxf/wotsit.org/dxf13/R13DXF/R13DXF.html)), if there is any better information please recommend.
I haven't written much English, so please forgive me if the writing is a little inappropriate.
thanks!