fougue/mayo #283: After removing the node in "libtree.h", only the node is marked as true, and the node content "data" is not completely removed.

comely13, 2024-07-30:
`template<typename T> void Tree<T>::removeRoot(TreeNodeId id)
{
    Expects(this->nodeIsRoot(id));

    // TODO Mark all deep children nodes as 'deleted'
    auto it = std::find(m_vecRoot.begin(), m_vecRoot.end(), id);
    if (it != m_vecRoot.end()) {
        TreeNode* node = this->ptrNode(id);
        Expects(node != nullptr);
        node->isDeleted = true;
        m_vecRoot.erase(it);
    }
}` In this part, if the model is opened repeatedly and then removed, the node content is not completely deleted, which will cause the memory to continue to increase.

HuguesDelorme, 2025-04-04:
I made some changes providing these fixes:
* Clear the model tree when count of root nodes shrinks down to 0
* Deep mark "deleted" all the children nodes when a root node is removed

First fix is nice to have in the typical use case where the document is reloaded after external change. In such case all "garbage" nodes are cleared
Second fix is here just for coherency. When a root node is removed, all the sub-tree is removed as well. By "remove" it means nodes are marked as "deleted" so this prevents those nodes to be further traversed.

There's a use-case which will lead to memory increase. If a document has multiple entities(ie root nodes), if the user remove one then the memory of entity sub-tree won't be released. Though that should be rather limited: the model tree stores `TDF_Label` which is basically a pointer.

comely13, 2025-04-04:
Thank you very much for your answer.

closed 2025-04-04 by comely13