fougue/mayo #285: Capping hatch scale is to small

Jopie01, 2024-07-30:
I've build Mayo myself and after clipping a model I see a complete black plane. I've also compared it with version 0.8.0 which I downloaded as AppImage. Below you can see the differences.

Development version I builded myself:
[image]

Version 0.8.0-x86_64.AppImage:
[image]

As you can see the light is different and the hatching is completely black. Has this something to do with OCC?

OS: Fedora 39
Platform: x86_64

HuguesDelorme, 2024-07-30:
Can you check your options in Mayo?
[image]

HuguesDelorme, 2024-07-30:
Please give also the OpenCasccade version you use to build Mayo

Jopie01, 2024-07-30:
I have also both 'Capping' and 'Capping Hatch' checked. If I uncheck the 'Capping Hatch' the colors are back.

> Please give also the OpenCasccade version you use to build Mayo

I followed the build instructions for Linux on https://github.com/fougue/mayo/wiki/Build-instructions-for-Linux and specifically Fedora. I have the following opencascade packages installed:
```
opencascade-foundation-7.6.3-4.fc39.x86_64
opencascade-modeling-7.6.3-4.fc39.x86_64
opencascade-ocaf-7.6.3-4.fc39.x86_64
opencascade-visualization-7.6.3-4.fc39.x86_64
opencascade-draw-7.6.3-4.fc39.x86_64
opencascade-devel-7.6.3-4.fc39.x86_64
```

HuguesDelorme, 2024-08-01:
Mayo AppImage is built with OpenCascade 7.7.0
Right now I have no idea what is causing the bug you reported. If you zoom in, do you see the capping hatch at some point?
I will have to install fedora, build mayo and try to reproduce the problem

Jopie01, 2024-08-01:
> Mayo AppImage is built with OpenCascade 7.7.0

Is it possible to build a new AppImage with the latest changes? Maybe there has been a change in occ version 7.6 and 7.7 which is causing this.

HuguesDelorme, 2024-08-01:
Latest Mayo is working fine on Ubuntu 20.04 with OpenCascade 7.7
[image]

AppImages are produced only with new release, right now there's no CI to automatically produce Mayo AppImages

Jopie01, 2024-08-01:
Well then I think it's because of an old version of opencascade. 

>  I will have to install fedora, build mayo and try to reproduce the problem

You have to install Fedora rawhide which comes with version 7.8. Before rawhide there is 7.6. Building opencascade from source is going to be a real pain. I will spin up a new VM with Fedora rawhide and see what it brings.

Jopie01, 2024-08-02:
I created a new VM and installed Fedora rawhide nightly from https://openqa.fedoraproject.org/nightlies.html. I choose the KDE Live iso and installed that. Then I followed the build instructions for building Mayo. In this case opencascade version 7.8 was used.

But the problem still exists 😢 .

HuguesDelorme, 2024-08-02:
Bad news, one thing that can be tested is to check clipping features in OpenCascade Draw(TCL) environment
This will help figure out if the issue is on Mayo or OpenCascade side

Jopie01, 2024-08-02:
The weird thing to me is that it fails on Fedora, but Ubuntu is fine. I don't know when a new release will be made, but I'm very interested to see what the AppImage of that release does. Maybe there is even a opencascade dependency which is failing etc. So I'm going to wait for the new release before testing and digging deeper.

Jopie01, 2024-08-13:
OK, problem is with Opencascade and I think with the RPM packages in Fedora. When I execute the commands, I get a 3D window with the clipped part with dimensions. But the hatching is completely black. What I noticed on the console was:

```
Error: no image library available
Error: no image library available
```
So I looked at the system information which showed the OCCT build configuration:
```
Open CASCADE Technology 7.6.3
Tk enabled (HAVE_TK)
Xlib enabled (HAVE_XLIB)
TBB enabled (HAVE_TBB)
FreeType enabled (HAVE_FREETYPE)
FreeImage disabled
FFmpeg disabled
OpenGL: enabled (HAVE_OPENGL_EXT)
OpenVR disabled
RapidJSON disabled
Draco disabled
VTK enabled (HAVE_VTK)
Exceptions enabled
Compiler: GCC 13.1.1
Architecture: AMD64
OS: Linux
```
It's clearly a problem in how Opencascade is build for Fedora.

HuguesDelorme, 2024-08-13:
Well spotted!
FreeImage was disabled when building OpenCascade RPM packages
Capping hatch uses a texture so it needs to load image files through FreeImage library

closed 2024-08-13 by HuguesDelorme