fougue/mayo #289: Feature request: export to image using active camera orientation and transparent background

virtualdj, 2024-08-04:
I would like the **Image** output feature of _File > Export selected items_ to obey the current camera view in the UI, if possible.
It would be great to have also a **transparent** background option when using formats that support it (such as PNG).

The _Save View to image_ feature is not the same, because it also outputs the grid and controls.

Thanks for this wonderful piece of software! 😎

HuguesDelorme, 2024-08-06:
Unfortunately, to my knowledge OpenCascade 3D view doesn't support transparent background

virtualdj, 2024-08-06:
OK, what a pity, it could have been useful to integrate into other apps... anyway, at least saving with same orientation could be enough 😃

virtualdj, 2024-08-06:
> Unfortunately, to my knowledge OpenCascade 3D view doesn't support transparent background

Hi, sorry if I reply again, but I found [this question](https://dev.opencascade.org/node/81222#comment-21046) on the OpenCascade forums.
I'm not really a C++ expert (actually I'm not a programmer at all), but could it be that changing this:
https://github.com/fougue/mayo/blob/3b4d7e6f67f6c49a0a777a6eaa00a2f21998d0e5/src/io_image/io_image.cpp#L191

to [`Graphic3d_BT_RGBA`](https://dev.opencascade.org/doc/refman/html/_graphic3d___buffer_type_8hxx.html#ac73b8613a40b278a1cbd43340ba93118) and providing the alpha component might work with PNG?

HuguesDelorme, 2024-08-07:
I tried this but it couldn't make it work:
* create a transparent PNG file
* call `V3d_View::SetBackgroundImage()` with this PNG file(within `ImageWriter::createV3dView()` function)
* set `dumpOptions.BufferType = Graphic3d_BT_RGBA;`(within `ImageWriter::createV3dView()` function)
* set `pixmap->SetFormat(Image_Format_RGBA);`(within `ImageWriter::createV3dView()` function)
* map `QImage::Format_RGBA8888` to `Image_Format_RGBA` in function `toQPixmap()`(within `qtgui_utils.cpp` file)

The problem certainly lies with `V3d_View` not supporting transparent background(no alpha color channel)

virtualdj, 2024-08-07:
> * create a transparent PNG file
> * call `V3d_View::SetBackgroundImage()` with this PNG file(within `ImageWriter::createV3dView()` function)

Why is it necessary to use `SetBackgroundImage()`? I would like the created image to be transparent, not to add a transparent image as a background. But I see now that `[V3d_View::SetBackgroundColor()](https://dev.opencascade.org/doc/refman/html/class_v3d___view.html#a1f0d95bbd63372624236ce3b91f0b463)` doesn't have options for RGBA 😢 so probably we're in trouble.