fougue/mayo #290: Mayo Conv doesn’t export IGES colors to OBJ/MTL

SamuelTallet, 2024-08-04:
Hello,

First of all, thank you for this great project.

**Issue**

Here is a colored IGES file opened in Mayo.

[image]

When I export to OBJ format, from the Mayo 0.8.0 app, this colored IGES file, OBJ and MTL files are created and the colors are preserved.

However, if I do the same op with the Mayo Conv CLI (ec5d7d7): only the OBJ file is created and the colors are lost. Here is a screenshot.

[image]

PS: This IGES file can be downloaded [here](https://www.cadcrowd.com/3d-models/snes-controller/snes-controller).

SamuelTallet, 2026-11-11:
It's OK with Mayo Conv 0.10.0, thank you!

closed 2026-11-11 by SamuelTallet

HuguesDelorme, 2026-11-11:
> It's OK with Mayo Conv 0.10.0, thank you!

Thanks for having checked the bug with latest release !
This was probably solved with update to OpenCascade 7.9.x