fougue/mayo #305: Having issues building Mayo and running your pre-built appimages

dCo3lh0, 2024-11-07:
I've been trying to build Mayo here, built with success, but I'm having this error at launching mayo, it instantly crashes when opening a file, it launches without issues, but when i try opening a file it crashes, here is the error, its refering to OpenGL.

```
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Segmentation fault
```

HuguesDelorme, 2024-11-07:
Hello, can you please post what's reported by _Help->System Information..._ dialog?
Use the _Copy to Clipboard_ button
AppImage is crashing as well?

dCo3lh0, 2024-11-08:
Here is what you asked:

```
Mayo: v0.8.0  commit:665e386  revnum:1227  64bit

OS: TUXEDO OS 4 [linux version 6.11.0-105009-tuxedo]
Current CPU Architecture: x86_64

Qt 5.15.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 5.3.1 20160406 (Red Hat 5.3.1-6)) on "xcb" 
    QStyle keys: Windows; Fusion
    Image formats(read): bmp cur gif icns ico jpeg jpg pbm pgm png ppm svg svgz tga tif tiff wbmp webp xbm xpm
    Image formats(write): bmp cur icns ico jpeg jpg pbm pgm png ppm tif tiff wbmp webp xbm xpm

OpenCascade: 7.7.0 (build)

Assimp: 5.3.0 rev:0 branch:? flags:shared|single-threaded

gmio: 0.4.1 (build)

Import(read) formats:
    DXF STEP IGES OCCBREP STL GLTF OBJ VRML OFF PLY AMF 3DS 3MF COLLADA FBX X3D Blender 
Export(write) formats:
    STEP IGES OCCBREP STL VRML GLTF OBJ OFF PLY AMF Image 

Locale:
    std::locale:
        name: pt_BR.UTF-8
        numpunct.decimal_point: , U+002C
        numpunct.thousands_sep: . U+002E
        numpunct.grouping: 3 
        numpunct.truename: true
        numpunct.falsename: false
    QLocale:
        name: pt_BR
        language: Portuguese
        measurementSytem: MetricSystem
        textDirection: LeftToRight
        decimalPoint: , U+002C
        groupSeparator: . U+002E

C++ StdLib:
    std::thread::hardware_concurrency: 16
    std::filepath::path::preferred_separator: / U+002F

OpenGL:
    GLXDirectRendering: Yes
    GLXVendor: SGI
    GLXVersion: 1.4
    GLXClientVendor: NVIDIA Corporation
    GLXClientVersion: 1.4
    GLvendor: NVIDIA Corporation
    GLdevice: NVIDIA GeForce RTX 2060/PCIe/SSE2
    GLversion: 4.6.0 NVIDIA 560.35.03
    GLSLversion: 4.60 NVIDIA
    Max texture size: 32768
    Max FBO dump size: 32768x32768
    Max combined texture units: 192
    Max MSAA samples: 32
    Viewport: 100x100
    Window buffer: RGB8 ALPHA0 DEPTH24 STENCIL0
    GPU free memory: 5066 MiB
    GPU memory: 6144 MiB
    MaxDegreeOfAnisotropy: 16
    MaxDrawBuffers: 8
    MaxClipPlanes: 8
    HasRayTracing: true
    HasRayTracingTextures: true
    HasRayTracingAdaptiveSampling: true
    UseVBO: true
    MaxDumpSizeX: 32768
[... 181 lines cut ...]
    SYSTEMD_EXEC_PID="260111"
    SYSTEMD_PAGER="less"
    TERM="linux"
    USER="dco3lh0"
    WAYLAND_DISPLAY="wayland-0"
    XAUTHORITY="/run/user/1000/xauth_tIvafo"
    XDG_ACTIVATION_TOKEN="not-granted-666"
    XDG_CONFIG_DIRS="/home/dco3lh0/.config/kdedefaults:/etc/xdg"
    XDG_CURRENT_DESKTOP="KDE"
    XDG_DATA_DIRS="/home/dco3lh0/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share"
    XDG_MENU_PREFIX="plasma-"
    XDG_RUNTIME_DIR="/run/user/1000"
    XDG_SEAT="seat0"
    XDG_SEAT_PATH="/org/freedesktop/DisplayManager/Seat0"
    XDG_SESSION_CLASS="user"
    XDG_SESSION_DESKTOP="KDE"
    XDG_SESSION_ID="3"
    XDG_SESSION_PATH="/org/freedesktop/DisplayManager/Session1"
    XDG_SESSION_TYPE="wayland"
    XDG_VTNR="1"
    XKB_DEFAULT_LAYOUT="us"
    XKB_DEFAULT_MODEL="pc86"
    XKB_DEFAULT_VARIANT="altgr-intl"

```

dCo3lh0, 2024-11-08:
after installing some assimp libraries here, the message in the terminal changed, now its this:

```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
WARNING: The X11 connection broke: No error (code 0)
XIO:  fatal IO error 0 (Success) on X server ":1"
      after 1450 requests (1450 known processed) with 17 events remaining.

```

HuguesDelorme, 2024-11-08:
Thanks for the infos
Can you set environment variable `QT_DEBUG_PLUGINS` to 1 and then run mayo?
````bash
export QT_DEBUG_PLUGINS=1
./Mayo-0.8.0-x86_64.AppImage
````

dCo3lh0, 2024-11-09:
here it is:

```
$ ./Mayo.AppImage
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforms" ...
DEBUG: QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforms/libqxcb.so"
WARNING: Found metadata in lib /tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

DEBUG: Got keys from plugin meta data ("xcb")
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/bin/platforms" ...
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforms/libqxcb.so"
DEBUG: loaded library "Xcursor"
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platformthemes" ...
DEBUG: QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platformthemes/libqxdgdesktopportal.so"
WARNING: Found metadata in lib /tmp/.mount_Mayo.AUo8tUA/usr/plugins/platformthemes/libqxdgdesktopportal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformThemeFactoryInterface.5.1",
    "MetaData": {
        "Keys": [
            "xdgdesktopportal",
            "flatpak",
            "snap"
        ]
    },
    "archreq": 0,
    "className": "QXdgDesktopPortalThemePlugin",
    "debug": false,
    "version": 331520
}

DEBUG: Got keys from plugin meta data ("xdgdesktopportal", "flatpak", "snap")
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/bin/platformthemes" ...
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforminputcontexts" ...
DEBUG: QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforminputcontexts/libcomposeplatforminputcontextplugin.so"
WARNING: Found metadata in lib /tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforminputcontexts/libcomposeplatforminputcontextplugin.so, metadata=
{
    "IID": "org.qt-project.Qt.QPlatformInputContextFactoryInterface.5.1",
    "MetaData": {
        "Keys": [
            "compose",
            "xim"
        ]
    },
    "archreq": 0,
    "className": "QComposePlatformInputContextPlugin",
    "debug": false,
    "version": 331520
}

DEBUG: Got keys from plugin meta data ("compose", "xim")
DEBUG: QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so"
WARNING: Found metadata in lib /tmp/.mount_Mayo.AUo8tUA/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so, metadata=
{
    "IID": "org.qt-project.Qt.QPlatformInputContextFactoryInterface.5.1",
    "MetaData": {
        "Keys": [
            "ibus"
        ]
[... 222 lines cut ...]
    "className": "QWebpPlugin",
    "debug": false,
    "version": 331520
}

DEBUG: Got keys from plugin meta data ("webp")
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/bin/imageformats" ...
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqsvg.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqgif.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqicns.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqico.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqjpeg.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqtga.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqtiff.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqwbmp.so"
DEBUG: loaded library "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/imageformats/libqwebp.so"
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/iconengines" ...
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/bin/iconengines" ...
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/plugins/accessible" ...
DEBUG: QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Mayo.AUo8tUA/usr/bin/accessible" ...
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Segmentation fault

```

HuguesDelorme, 2024-11-11:
QT_DEBUG_PLUGINS doesn't help spot the problem.
I need to install TUXEDO OS on VM machine and try to reproduce the crash

HuguesDelorme, 2024-11-12:
`Mayo-0.8.0-x86_64.AppImage` is working fine on TUXEDO OS 4 in VirtualBox machine:
[image]

dCo3lh0, 2024-11-12:
I don't know if it will help, but did you run on x11 or wayland, I'm on Wayland now, but running on xWayland mode that bridges display to x11, forgot to add this information, very late to inform that, sorry.

dCo3lh0, 2024-11-12:
This totally has to do with OpenCascade Library, i've installed the occt-misc package to add all the libraries, so no libraries missing here. but maybe it has problems with dealing with wayland.

dCo3lh0, 2024-11-12:
now its displaying me this here:

```
DEBUG: OpenCascade settings file doesn't exist or is not readable [path=opencascade.conf]
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
WARNING: The X11 connection broke: No error (code 0)
X connection to :1 broken (explicit kill or server shutdown).

```

dCo3lh0, 2024-11-12:
I have setup here NVIDIA-PRIME, running a custom command made by tuxedo to made offloading the GPU command easier, the opencascade.conf part of the error disappears, the command is "dgpu-run ./Mayo.AppImage"

dCo3lh0, 2024-11-12:
solved! I've looked at this error issue here relating to wayland issue #178 and it pointed me to try force OpenGL fallback as said on this other issue here #180 widget option, i've configured, closed and reopened and voila, its now working.

dCo3lh0, 2024-11-12:
the error is still the same but the program doesn't crash

dCo3lh0, 2024-11-12:
damn, it keeps happening, if i open a new document it displays the viewport, if i try to load any object or mesh, it crashes and gives me segmentation fault, i give up.

HuguesDelorme, 2024-11-13:
Good to know Mayo is running on your machine at last!
So the solution was just about enabling the OpenGL fallback widget?
Concerning support of Blender file format, this is provided by [assimp](https://github.com/assimp/assimp) plugin
Some note in this [page](https://github.com/assimp/assimp/blob/master/doc/Fileformats.md) mentions:
_support for [BLEND](https://en.wikipedia.org/wiki/.blend_(file_format)) is deprecated. It is too time-consuming to maintain an undocumented format which contains so much more than we need_
This probably explains the crash you get when attempting to load a blend file. Support of this format should be removed in a future Mayo version

dCo3lh0, 2024-11-14:
just a last question. If by fallbacking to OpenGL solves this, what is the default graphical library that is used by Mayo? Just curious... it is OpenCascade? I'm a little newbie on this, and this library has has issues with Wayland as it seems and OpenGL works fine.

HuguesDelorme, 2024-11-14:
> just a last question. If by fallbacking to OpenGL solves this, what is the default graphical library that is used by Mayo? Just curious... it is OpenCascade? I'm a little newbie on this, and this library has has issues with Wayland as it seems and OpenGL works fine.

Mayo uses OpenCascade for 3D rendering, and this library is based on OpenGL.
There are two different ways in Mayo to integrate OpenCascade's 3D rendering with a Qt window : `QOpenGLWidget` and `QWidget`

1) Since OpenCascade 7.6 version it's possible to use `QOpenGLWidget` with OpenCascade. This allows widgets with translucid background to be correctly displayed over `V3d_View`. OpenCascade/Qt integration is done through the OpenGL framebuffer object created by Qt.
[image]

2) The "legacy" way is to use a `QWidget` and in this case OpenCascade/Qt integration is made with a dedicated implementation of OpenCascade's `Aspect_Window` around `QWidget`. Widgets with tranlucid backrounds are not supported
[image]

The first solution is nice but may have issues on Linux and macOS(works nice on Windows).
The second solution is less fancy but has proved to be robust

Wayland is not supported by OpenCascade, that's a limitation on Linux systems

dCo3lh0, 2024-11-14:
Thats all i needed to know, thanks for all!, I'm gonna close with this comment.

dCo3lh0, 2024-11-14:
at least, it seems somebody, is working on it here:
https://github.com/Escain/OCCT_Wayland

closed 2024-11-14 by dCo3lh0