fougue/mayo #306: Unable to build mayo 0.8.0 on Ubuntu 24.10

jhonny-oliveira, 2024-11-20:
Hi!

I'm unable to build mayo 0.8.0 on Ubuntu 24.10. This is the error:

```In file included from src/base/caf_utils.cpp:7:
src/base/caf_utils.h:47:10: fatal error: TDF_LabelMapHasher.hxx: No such file or directory
   47 | #include <TDF_LabelMapHasher.hxx>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
"TDF_LabelMapHasher.hxx" is present as part of libocct-ocaf-dev (7.6.3) on older Ubuntu versions (<= 24.04), but not on the most recent version: libocct-ocaf-dev (7.8.1).

Can you please advise on the best workaround/patch?

Thank you!
Jhonny Oliveira @ https://xtradeb.net

Ref.:
https://packages.ubuntu.com/noble/amd64/libocct-ocaf-dev/filelist
https://packages.ubuntu.com/oracular/amd64/libocct-ocaf-dev/filelist
https://launchpad.net/~xtradeb/+archive/ubuntu/apps/+packages?field.name_filter=mayo

HuguesDelorme, 2024-11-20:
Hello Jhonny,

Mayo v0.8.0 source code is compatible with OpenCascade v7.3 up to v7.7
Upcoming version v0.9.0 source code was made compatible with latest OpenCascade v7.8
You can use the relevant changes from this commit to locally patch Mayo v0.8.0 source code:
https://github.com/fougue/mayo/commit/d1b3545e9257b3bba26b3c2805b510af5d163048

jhonny-oliveira, 2024-11-23:
I also had to apply commit c4fdeb2 to get it working.

Thank you!

closed 2024-11-23 by jhonny-oliveira